roboticslab-uc3m/yarp-devices #162: Derive from concrete YARP classes instead of providing custom implementation

PeterBowman, 2018-01-06:
YARP's fakeMotionControl device inherits from several `ImplementX` concrete classes, e.g. `ImplementVelocityControl2`, a ready-to-use implementation of `IVelocityControl2`. See inheritance diagram in [fakeMotionControl's docs](http://www.yarp.it/classyarp_1_1dev_1_1FakeMotionControl.html). I wonder if we could use this in our CAN devices to spare a few lines of code.

jgvictores, 2018-02-27:
Blocking #171

PeterBowman, 2018-07-19:
Note to self: take a look at http://www.yarp.it/classyarp_1_1dev_1_1ControlBoardHelper.html (perhaps for internal use only).

jgvictores, 2018-12-18:
Does not really block #171

PeterBowman, 2018-12-18:
This WIP led to https://github.com/robotology/yarp/pull/1867, another PR is coming.

jgvictores, 2018-12-21:
Blocks #171.

PeterBowman, 2019-05-16:
Might render invalid due to https://github.com/roboticslab-uc3m/yarp-devices/issues/211.

PeterBowman, 2019-05-16:
For instance, if we take a look at [ImplementControlLimits.h](https://github.com/robotology/yarp/blob/v3.1.1/src/libYARP_dev/include/yarp/dev/ImplementControlLimits.h), there are two interfaces:

* `StubImplControlLimitsRaw`: targets a `Raw` device, provides a stub implementation that prints an error notice and returns `false`. The only place I could use these interfaces is CuiAbsolute, leaving here my abandoned commit for future reference: https://github.com/roboticslab-uc3m/yarp-devices/commit/cc80172b4a56564f63d5aa8cfe85821f56b96ee2. Superseded by https://github.com/roboticslab-uc3m/yarp-devices/issues/211.

* `ImplementControlLimits`: comprises an cpp counterpart ([ImplementControlLimits.cpp](https://github.com/robotology/yarp/blob/v3.1.1/src/libYARP_dev/src/ImplementControlLimits.cpp)) that performs basic mapping and call forwarding to raw devices assuming a correct initialization. Could be implemented in:
  * `FakeControlboard`: probably going to be removed with https://github.com/roboticslab-uc3m/yarp-devices/issues/192.
  * `CanBusControlboard`: currently performs simple routing, but I have a major rework in mind involving CAN reads&writes (https://github.com/roboticslab-uc3m/yarp-devices/issues/209). Would require implementing currently omitted methods in the raw device implementations.

The `CanBusControlboard` thing could be revisited in the future if https://github.com/roboticslab-uc3m/yarp-devices/issues/209 cannot be done. Closing for now as WONTFIX.

closed 2022-06-04 by PeterBowman