roboticslab-uc3m/yarp-devices #164: Setting color debug for yarp-devices

rsantos88, 2018-01-29:
I remember that we could activate and deactivate the color debug option, selecting what **type** of messages we would want to see, for example when we were debugging some messages running `launchManipulation` of `launchLocomotion` and we don't want to see all the stream of messages (info, warnings, successful ... etc)
Have we lost that function doing ccmake..?
In my case, when I do `ccmake` on `yarp-devices` I can't find it:
```bash
 CMAKE_BUILD_TYPE                 Release                                                                                                                                                                   
 CMAKE_INSTALL_PREFIX             /usr/local                                                                                                                                                                
 COLOR_DEBUG_DIR                  /usr/local/lib/COLOR_DEBUG/cmake                                                                                                                                          
 ENABLE_AravisGigE                OFF                                                                                                                                                                       
 ENABLE_CanBusControlboard        ON                                                                                                                                                                        
 ENABLE_CanBusHico                ON                                                                                                                                                                        
 ENABLE_CuiAbsolute               ON                                                                                                                                                                        
 ENABLE_FakeControlboard          ON                                                                                                                                                                        
 ENABLE_FakeJoint                 ON                                                                                                                                                                        
 ENABLE_Jr3                       ON                                                                                                                                                                        
 ENABLE_LacqueyFetch              ON                                                                                                                                                                        
 ENABLE_LeapMotionSensor          OFF                                                                                                                                                                       
 ENABLE_OneCanBusOneWrapper       ON                                                                                                                                                                        
 ENABLE_ProximitySensorsClient    ON                                                                                                                                                                        
 ENABLE_SpaceNavigator            OFF                                                                                                                                                                       
 ENABLE_TechnosoftIpos            ON                                                                                                                                                                        
 ENABLE_TextilesHand              ON                                                                                                                                                                        
 ENABLE_TwoCanBusThreeWrappers    ON                                                                                                                                                                        
 ENABLE_WiimoteSensor             OFF                                                                                                                                                                       
 ENABLE_YarpPlugins               ON                                                                                                                                                                        

CMAKE_BUILD_TYPE: Choose the type of build, options are: Debug Release MinSizeRel RelWithDebInfo                                                                                                             
Press [enter] to edit option                                                                                                                                                              CMake Version 3.5.1
Press [c] to configure
Press [h] for help           Press [q] to quit without generating
Press [t] to toggle advanced mode (Currently Off)
```
```bash
                                                     Page 2 of 2
 ENABLE_checkCanBus               ON                                                                                                                                                                        
 ENABLE_coverage                  OFF                                                                                                                                                                       
 ENABLE_dumpCanBus                ON                                                                                                                                                                        
 ENABLE_grabbercontrols2gui       OFF                                                                                                                                                                       
 ENABLE_launchLocomotion          ON                                                                                                                                                                        
 ENABLE_launchManipulation        ON                                                                                                                                                                        
 ENABLE_oneCanBusOneWrapper       ON                                                                                                                                                                        
 ENABLE_tests                     ON                                                                                                                                                                        
 Leap_LIBRARY                     Leap_LIBRARY-NOTFOUND                                                                                                                                                     
 Leap_ROOT_DIR                    Leap_ROOT_DIR-NOTFOUND                                                                                                                                                    
 SPNAV_INCLUDE_DIR                SPNAV_INCLUDE_DIR-NOTFOUND                                                                                                                                                
 SPNAV_LIBRARY                    SPNAV_LIBRARY-NOTFOUND                                                                                                                                                    
 XWiimote_INCLUDE_DIR             XWiimote_INCLUDE_DIR-NOTFOUND                                                                                                                                             
 XWiimote_LIBRARY                 XWiimote_LIBRARY-NOTFOUND                                                                                                                                                 
 YARP_DIR                         /usr/local/lib/YARP                                                                                                                                                       
 YCM_DIR                          /usr/local/share/YCM/cmake                                                                                                                                                
[... cut ...]
ENABLE_checkCanBus: Enable/disable checkCanBus program                                                                                                                                                       
Press [enter] to edit option                                                                                                                                                              CMake Version 3.5.1
Press [c] to configure
Press [h] for help           Press [q] to quit without generating
Press [t] to toggle advanced mode (Currently Off)
```

PeterBowman, 2018-01-29:
Are you pulling CD through YCM (i.e. it's not installed on your system)? If not, `git pull` and reinstall CD. Otherwise, press `[c]` again, you should see the missing cache options, then. Or even better, delete the cache and issue `cmake ..`, you'll see a disclaimer message regarding CD settings.

https://github.com/roboticslab-uc3m/yarp-devices/blob/99c1062747a40d7336b3b21b98660cd1d86f10ee/cmake/BuildCOLOR_DEBUG.cmake#L12-L20

PeterBowman, 2018-01-29:
BTW: CD options won't show up unless you enable advanced mode.

`Press [t] to toggle advanced mode (Currently Off)`

rsantos88, 2018-01-29:
I get it now. The trick is pressing [t] (togglle advanced mode) and we can see this options..

jgvictores, 2018-01-29:
Closing issue, started https://github.com/roboticslab-uc3m/color-debug/issues/7 as follow-up.

closed 2018-01-29 by jgvictores