roboticslab-uc3m/yarp-devices #165: Handle 3rd-party deps for PhidgetSpatial device

PeterBowman, 2018-02-04:
As a follow-up to #163, there should be a clean way to enable the `PhidgetSpatial` device and let CMake tell us whether required deps are available or not (just don't ignore that and make it crash on compile time). Necessary headers and libs are currently located in [`extern/phidget/`](https://github.com/roboticslab-uc3m/yarp-devices/tree/6dcce3b235dab788bda1b3425207b2d55b6cb074/extern/phidget). Ideas:

* If enabled, instruct CMake to unzip the TAR file and fetch its contents into the build process. Warn the user and disable the device otherwise.
* Write a `FindSpatial.cmake` module and proceed as usual (keep in mind: roboticslab-uc3m/QA#18).

PeterBowman, 2018-02-04:
FIXME (`cmake-opts-on` branch):

https://github.com/roboticslab-uc3m/yarp-devices/blob/81df04907fcf5e6303d627a6b4570b64b9cfd28b/libraries/YarpPlugins/PhidgetSpatial/CMakeLists.txt#L8

jgvictores, 2018-02-04:
> * Write a `FindSpatial.cmake` module and proceed as usual (keep in mind: roboticslab-uc3m/QA#18).

+1 to this option. Additionally, seems like option that can provide better support for different OS.

jgvictores, 2018-02-06:
Related: #167

PeterBowman, 2018-02-06:
Done at 37f39f8 (find module + docs for PhidgetSpatial device + build coverage on Travis).

closed 2018-02-06 by PeterBowman