roboticslab-uc3m/yarp-devices #167: Clean extern

jgvictores, 2018-02-06:
Clean [extern](https://github.com/roboticslab-uc3m/yarp-devices/tree/develop/extern) ([permalink](https://github.com/roboticslab-uc3m/yarp-devices/tree/f93b645e228a38e60f48ec49d5757b87fbd66e8d/extern)):
1. hcanpci: obsolete
1. phidget: can be put in https://sourceforge.net/projects/roboticslab/files/External/

PeterBowman, 2018-02-23:
> phidget: can be put in https://sourceforge.net/projects/roboticslab/files/External/

Note: update [.travis.yml](https://github.com/roboticslab-uc3m/yarp-devices/blob/b80bbb576721721e1f9b418873f4387597b73e57/.travis.yml#L25-L32) to point to the new file location.

PeterBowman, 2023-03-26:
Done at https://github.com/roboticslab-uc3m/yarp-devices/commit/7d6b634315e7daf5ade69546d0dfecbf5300d2ed. The old stuff has been moved to https://sourceforge.net/projects/roboticslab/files/External/.

closed 2023-03-26 by PeterBowman