roboticslab-uc3m/yarp-devices #169: Decouple GLib package search from ARAVIS find module

PeterBowman, 2018-02-08:
There is a `find_package(GLib REQUIRED)` line in current [`FindARAVIS.cmake`](https://github.com/roboticslab-uc3m/yarp-devices/blob/8f0234a5cf02ddb9a0739dd0c670f794ed0894a4/cmake/find-modules/FindARAVIS.cmake#L8) that breaks the CMake dependency mechanism if `libglib2.0-dev` is not installed (roboticslab-uc3m/questions-and-answers#18).

From https://github.com/roboticslab-uc3m/questions-and-answers/issues/18#issuecomment-364137279:

> Looking at [these lines](https://github.com/roboticslab-uc3m/yarp-devices/blob/f92e09a218e906b014e009ddb3831fc6a07233df/libraries/YarpPlugins/AravisGigE/CMakeLists.txt#L5-L18), cmake always goes through `find_package(ARAVIS QUIET)`. Now, if we look at [FindARAVIS.cmake](https://github.com/roboticslab-uc3m/yarp-devices/blob/f92e09a218e906b014e009ddb3831fc6a07233df/cmake/find-modules/FindARAVIS.cmake#L8), it has a `find_package(GLib REQUIRED)` which makes everything break, always, even if we are not using `AravisGigE`... :-(

Proposal at https://github.com/roboticslab-uc3m/questions-and-answers/issues/18#issuecomment-364140187:

> That `find_package(GLib REQUIRED)` should be moved to the caller list file (e.g. `AravisGigE/CMakeLists.txt`). Similarly, you have to explicitly call both `find_package(OpenRAVE REQUIRED)` and `find_package(Boost REQUIRED)` in the openrave-yarp-plugins repo.
>
> However, I'd personally give a try to the `find_dependency()` command ([ref](https://cmake.org/cmake/help/v3.0/module/CMakeFindDependencyMacro.html)) within `FindARAVIS.cmake`. I used to have one at [ROBOTICSLAB_KINEMATICS_DYNAMICSConfig.cmake.in](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/9112f27eff6f52f841bfd06802be777d2d9e4a89/cmake/template/ROBOTICSLAB_KINEMATICS_DYNAMICSConfig.cmake.in#L40-L43).

David-Estevez, 2018-02-08:
`GLib` is not a dependency of our project, but a dependency of `Aravis`. Therefore, it seems intuitive to me to look for it within the file that looks for `Aravis`.

As an example, imagine we need `OpenCV` as a dependency for our project. `OpenCV` has several dependencies that have to be installed for it to work, but as a user you don't need to keep track of all of them (it would somehow break encapsulation), you just look for `OpenCV` and that file looks for all the dependencies required. 

 Is there any chance of looking for `GLib` it in `FindARAVIS.cmake` and not breaking the dependency mechanism?

PeterBowman, 2018-02-08:
> Is there any chance of looking for GLib it in FindARAVIS.cmake and not breaking the dependency mechanism?

Yes, I aim to solve it with `find_dependency()` called from within `FindARAVIS.cmake`, thus encapsulation should be preserved (i.e. Glib not found => `find_package(ARAVIS)` reports a failure).

> OpenCV has several dependencies that have to be installed for it to work, but as a user you don't need to keep track of all of them

We speak about transitive binary dependencies here, i.e. OpenCV *.so files that depend on other 3rd-party *.so files. You'll probably need to install required apt packages anyway (cf. [installing YARP from sources](http://www.yarp.it/install_yarp_linux.html#install_linux_from_sources)), which is nicer because CMake code is not polluted, but this is the usual solution in respect of packaging, unless you prefer static linking.

It *is* a bit different once your library of choice imposes restrictions in their public interface, that is, in order to *use* the library, you need the development headers of another library (apart from *.so files or other binaries). This actually happens in the openrave-yarp-plugins: since OpenRAVE headers include Boost headers, the build system requires full knowledge about how to find paths/files for both packages. In that case, you call `find_package(Boost)` next to `find_package(OpenRAVE)`.

PeterBowman, 2018-02-09:
The `find_dependency()` solution works, see [Travis build](https://travis-ci.org/roboticslab-uc3m/yarp-devices/builds/339606134). Compare with current develop **without glib**: [ref](https://travis-ci.org/roboticslab-uc3m/yarp-devices/builds/339609953).

As a side note, @jgvictores observed that the `REQUIRED` parameter in `find_package(pkg REQUIRED)` is quite unusual, so we shouldn't worry about things breaking due to missing nested (chained?) dependencies.

PeterBowman, 2018-02-09:
Done at bc07402.

closed 2018-02-09 by PeterBowman

jgvictores, 2018-02-09:
For the record, in `/usr/share/cmake-3.5/Modules`, the command:
```bash
grep -ri "find_package(" * | grep REQUIRED | grep -v \#
```
only returns 3 to 5 results out of +350 cmake modules:
```
FindBLAS.cmake:    find_package(Threads REQUIRED)
FindCUDA.cmake:    find_package(Threads REQUIRED)
FindLAPACK.cmake:  find_package(BLAS REQUIRED)
FindLAPACK.cmake:      find_package(Threads REQUIRED)
FindXalanC.cmake:find_package(XercesC ${XalanC_XERCESC_REQUIRED})
```