roboticslab-uc3m/yarp-devices #172: adaptation of the new hand board

jgvictores, 2018-03-13:
_From @jmgarciah on March 13, 2018 10:2_

Once the PCB for the hand is developed. It is necessary to adapt the old code of the PIC to the new MBED.

_Copied from original issue: roboticslab-uc3m/teo-hardware-issues#19_

rsantos88, 2018-03-15:
The first step is to check the communications between the MBED and the computer. I am trying to send or receive messages over the CAN.
Steps taken until now:
* First, I've checked the connections. I've taken this model to revise the connections between the pins 29-30, and the transceiver. The same between the transceiver and the can bus connector:
[image]
I detected a bad welding and @jmgarciah revised and re-welded again the transceiver, because there was bad contact with the pins and the board.

* Once I revised the connections, now I am trying to know if there is a communication. I did a simple program that allow us to send messages from the `mbed` to manipulation computer: 
```c
#include "mbed.h"

Ticker ticker;
Serial pc(USBTX, USBRX);   // Serial connection
DigitalOut live_led(LED4); // CAN read toggle led (activity)
DigitalOut led1(LED1);     // sended
DigitalOut led2(LED2);     // received
CAN can(p30, p29);         // tx,rx
char counter = 0;

DigitalOut myled(LED1);

void liveled()
// Toggles the led
{
    live_led=!live_led;
}

void send() {
    printf("send()\n");
    // create the message
    CANMessage messageOut;
     
    messageOut.format = CANStandard; // or  CANExtended; standard or extended ID (can be skipped for standard)
    messageOut.id = 10;
    messageOut.len = 1; // length in bytes (1 to 8);
    messageOut.data[0] = counter; // repeat for each byte.    
     
    // send the message   
    if(can.write(messageOut)) {        
        printf("Message sent: %d\n", messageOut.data[0]);    
    }    
    
    counter++;    
    led1 = !led1;
}

void receive() {
    CANMessage msg;
        if(can.read(msg)) {
            pc.printf("Message received: %d\n", msg.data[0]);
            led2 = !led2;
        } 
}

int main() {
    pc.baud(115200);       // configure PC baudrate for USB
    pc.printf("main()\n");     
    // set up the can bus
    can.frequency(1000000);     // 1Mbit/s (the same as the drivers configuration)
    can.reset();        
    ticker.attach(&liveled, 1); // Send every 1 sec (to know that the mbed is running at this point)
    while(1) {
        send();
        wait(0.2);    
    }
}

```

With this program we can send packets every 200ms and with `dumpCanBus` running on manipulation-pc we should see the messages sent but there is no message captured .
* In other way, you can modify easily the code to call the `receive()` function within the loop instead `send()` function and do the same process but running `launchManipulation` on manipulation computer and receiving some can message through `mbed`. To verify this, I've connected the `mbed` by USB to my computer to monitoring the arrived messages using a serial connection program like [moserial](https://wiki.gnome.org/moserial) but no messages received over the `mbed`.

rsantos88, 2018-04-03:
The mbed firmware for the lacqueyFetch is finished, documented and working:
MBED firmware: https://os.mbed.com/users/raulsr988/code/LacqueyFetch/
PR of yarp-devices changes: https://github.com/roboticslab-uc3m/yarp-devices/pull/179

jgvictores, 2018-04-04:
Cool, thanks a lot!

Feel free to close this issue.

jgvictores, 2018-04-04:
PS: Copied https://os.mbed.com/users/raulsr988/code/LacqueyFetch/file/8ebb11ba21be/main.cpp to [yarp-devices/firmware/LacqueyFetch/main.cpp](https://github.com/roboticslab-uc3m/yarp-devices/blob/f857751e5b4695115d25eef9a50252eb870c2261/firmware/LacqueyFetch/main.cpp).

closed 2018-04-05 by rsantos88