roboticslab-uc3m/yarp-devices #184: Look at struct-int-to-double branch

jgvictores, 2018-05-17:
https://github.com/roboticslab-uc3m/yarp-devices/commit/184264ffc5b7b79b6ff26c8686a112f833e071c7

PeterBowman, 2019-07-29:
https://github.com/roboticslab-uc3m/yarp-devices/issues/203#issuecomment-494062739:

> Kinda unrelated, but I noticed that Clang complains about these lines:
>
> https://github.com/roboticslab-uc3m/yarp-devices/blob/a538b382ddec2649c7c1f1ea9400be867ac49170/libraries/YarpPlugins/Jr3/DeviceDriverImpl.cpp#L24-L26
>
> Each `5.5` value is narrowed down to `5` in order to fit into an integer-type variable.

jgvictores, 2019-07-30:
On a side note, we never mention it, but the device is tightly related to https://github.com/roboticslab-uc3m/jr3pci-linux

jgvictores, 2019-09-06:
> On a side note, we never mention it, but the device is tightly related to https://github.com/roboticslab-uc3m/jr3pci-linux

Ref at https://github.com/roboticslab-uc3m/yarp-devices/commit/380480966fd16bed8c279c6fc32d241799bf9f08