roboticslab-uc3m/yarp-devices #187: Implement ICurrentControl

AlvaroMartinezR, 2018-07-05:
When trying to diagnose the problems with the gait of TEO, @munozyanez and me encountered the new  functions of YARP for current control. This is a reminder for @jgvictores

AlvaroMartinezR, 2018-07-05:
Even though we have the getTorque functions, @jgvictores thought it was necessary. Documentation:
http://www.yarp.it/ICurrentControl_8h_source.html

jgvictores, 2018-07-05:
Renamed as `Think about`, adding low priority status. Similar to #157

PeterBowman, 2018-07-19:
Loosely related: the [`yarp::dev::ITorqueControl` interface](http://www.yarp.it/classyarp_1_1dev_1_1ITorqueControl.html) provides access to internal motor parameters ([setter](http://www.yarp.it/classyarp_1_1dev_1_1ITorqueControl.html#a0caee323b7bc948769c6792ae0e82525), [getter](http://www.yarp.it/classyarp_1_1dev_1_1ITorqueControl.html#a404b4775c156b7075c76584be24b0610)). Available: `bemf` ([back electromotive force](https://en.wikipedia.org/wiki/Counter-electromotive_force)), `bemf_scale`, `ktau`, `ktau_scale` (via [`yarp::dev::MotorTorqueParameters`](http://www.yarp.it/classyarp_1_1dev_1_1MotorTorqueParameters.html)).

PeterBowman, 2019-06-27:
Note: we set torques via `setRefTorque` (conveniently expressed in terms of motor current [here](https://github.com/roboticslab-uc3m/yarp-devices/blob/fc417a504e1143f5f65e3e4d36027105fc2021a6/libraries/YarpPlugins/TechnosoftIpos/ITorqueControlRawImpl.cpp#L47)), but `getTorque` reads motor current ([ref](https://github.com/roboticslab-uc3m/yarp-devices/blob/fc417a504e1143f5f65e3e4d36027105fc2021a6/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L321)).

jgvictores, 2019-06-27:
> Note: we set torques via setRefTorque (conveniently expressed in terms of motor current here), but getTorque reads motor current (ref).

True. I was aware of this fact but never documented it, sorry. ^^
We can estimate the torque via the motor constant `K`. On a side note, I recall on the iCub they had a separate module based on iKin/iDyn to

PeterBowman, 2019-07-09:
Note to self: split torque and current commands added at https://github.com/roboticslab-uc3m/yarp-devices/commit/89e22a7a896ce48ea82dd3791ab94b08b39e0a2f.

PeterBowman, 2019-07-09:
On start, frontal leg joints move sidewards. This is caused by a 0.5 A offset configured in the current controller, intended for gravity compensation. Such value can be accessed in EasySetup, drive setup window, current controller section, "Tune" option. This setting needs to be disabled.

PeterBowman, 2019-07-13:
Another note to self: query the iPOS drive for its peak current on device start. This value is immutable, drive-dependent and is involved in the conversion from and to internal current units.

PeterBowman, 2019-07-17:
All done at https://github.com/roboticslab-uc3m/yarp-devices/commit/0484b8a1712d0eb33ed3bfc6ba3160260c749896.

closed 2019-07-17 by PeterBowman

jgvictores, 2019-07-18:
@PeterBowman Thanks!!!