roboticslab-uc3m/yarp-devices #188: Implement IControlLimits(2)::setVelLimitsRaw in TechnosoftIpos

PeterBowman, 2018-07-19:
Currently unimplemented ([ref](https://github.com/roboticslab-uc3m/yarp-devices/blob/ec902e4d92faafa0fd71618e2393a3a4c5037122/libraries/YarpPlugins/TechnosoftIpos/IControlLimits2RawImpl.cpp#L86-L90)). Once done, remember to call this setter from CanBusControlboard:

https://github.com/roboticslab-uc3m/yarp-devices/blob/ec902e4d92faafa0fd71618e2393a3a4c5037122/libraries/YarpPlugins/CanBusControlboard/DeviceDriverImpl.cpp#L163-L170

PeterBowman, 2018-07-19:
Note: consume these values in our `ICartesianControl` implementations at kin-dyn and no longer support an angular velocity limit configuration parameter ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/ICartesianControl.h#L109)).

PeterBowman, 2019-03-12:
Make sure these limits will be taken into account in the IPositionDirect mode. In https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/159, a simple formula is proposed: delta_q/duration (where duration = ptModeMs).

jgvictores, 2019-03-12:
> Make sure these limits will be taken into account in the IPositionDirect mode. In [roboticslab-uc3m/kinematics-dynamics#159](https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/159), a simple formula is proposed: delta_q/duration (where duration = ptModeMs).

[#173](https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173#issuecomment-471958616), right?

PeterBowman, 2019-03-12:
Thanks, my bad, that link was meant to point at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173#issuecomment-471958616.

PeterBowman, 2019-05-05:
If I'm correct, the iPOS drive does not check whether the motor exceeds a software-imposed speed value (as it does with positions and currents). So, perhaps, this should be done by us - either by introducing another thread in the background, or by listening to incoming event messages (https://github.com/roboticslab-uc3m/yarp-devices/issues/159#issuecomment-488003042) - through frequent `IEncoders::getEncoderSpeed` calls (https://github.com/roboticslab-uc3m/yarp-devices/issues/189).

PeterBowman, 2019-05-16:
We were only allowed to **get** velocity limits as parsed from the corresponding .ini files. This is useful for user applications, see https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173. Implementing the **set** counterpart was trivial and has been done at https://github.com/roboticslab-uc3m/yarp-devices/commit/89cbdbad25320555e98d3ed7815eb4fef73669e7.

PeterBowman, 2019-05-16:
> Once done, remember to call this setter from CanBusControlboard

Not doing. Reference speeds&accs as well as position&velocity limits are passed to each TechnosoftIpos instance and stored on device config. Then, CAN messages must be sent to the drives in order to load these values on each slave. There is no max-speed object in the CAN dictionary we can act upon, therefore a `setVelLimits` instruction as suggested by myself would be useless.

PeterBowman, 2019-05-16:
Now, let's actually use these minVel/maxVel values! Should I ensure that reference speeds (refSpeed via .ini) are always equal or lower? Are there any other situations that would merit a velocity check (already dropped a note at https://github.com/roboticslab-uc3m/yarp-devices/issues/208)?

PeterBowman, 2019-05-18:
> Note: consume these values in our `ICartesianControl` implementations at kin-dyn and no longer support an angular velocity limit configuration parameter ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/ICartesianControl.h#L109), [roboticslab-uc3m/kinematics-dynamics#121 (comment)](https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/121#issuecomment-406344347)).

Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/a3600f79b637b9b63de6fda66fd0dd0a32a889b9 and https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/89020afa781ec6d57db4ab911414f2d4014207cc.

PeterBowman, 2019-05-20:
ASWJ the `minVels` property would be removed and interpreted on the YARP-side (i.e. via `IControlLimits` getter and setter) as the negative of `maxVels`.

PeterBowman, 2019-05-21:
Done at https://github.com/roboticslab-uc3m/yarp-devices/commit/d7773a1d35b45f2d4cae1a3850bd3e705db2a4db, velocity limit checks are performed on device configuration, in `IPositionControlRaw::setRefSpeedRaw` and `IVelocityMoveRaw::velocityMoveRaw`.

closed 2019-05-21 by PeterBowman

PeterBowman, 2019-05-25:
Added in PT mode at https://github.com/roboticslab-uc3m/yarp-devices/commit/7cac9f4752593a23c2c8db04a272fd25b9189a93 (experimental branch).