roboticslab-uc3m/yarp-devices #198: Improvements in IPositionDirect

PeterBowman, 2018-12-18:
Current (undesired/faulty) behavior:

- [ ] Change modes `IPositionControl -> IPositionDirect -> IPositionControl` and try to `positionMove()`: need to send this command twice, no response at first try.
- [ ] Change modes and move to a certain position after each step, `IPositionControl -> IPositionDirect -> IPositionControl -> IPositionDirect`. We won't be able to start the PT mode since `ptPointCounter` is never set to zero:
  https://github.com/roboticslab-uc3m/yarp-devices/blob/87b350cf1c678f649b6c88a3677c7d34e629c3d5/libraries/YarpPlugins/TechnosoftIpos/IPositionDirectRawImpl.cpp#L46-L55
- [ ] Once a buffer-full state is reached, new positions cannot be sent until the buffer is emptied (check mutex):
  https://github.com/roboticslab-uc3m/yarp-devices/blob/87b350cf1c678f649b6c88a3677c7d34e629c3d5/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L255-L266

jgvictores, 2018-12-18:
Related? "[TechnosoftIpos] Cannot change control mode some/many times #170"

PeterBowman, 2018-12-18:
Can't really tell unless we determine the underlying cause. Even though it's not stated explicitly in the title, #170 originated when trying to perform `[torq]` control after one or more mode transitions. I'd keep these two tickets along eachother since a different sequence of steps is described to reproduce their symptoms. Even if this results in a duplicate somewhere, we can point to the same problem from two distinct directions and resolve both at once.

rsantos88, 2019-03-11:
I've doing different tests, changing the mode in different occasions, and executing the position direct example (2ebc388fb44440b09400901a7ee4e2457365bfa3) that I developed. The result has been this:

[image]

**Screen output:**
launchManipulation -> [issue-change-mode.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/2952405/issue-change-mode.txt.zip)
examplePositionDirect -> [examplePositionDirect.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/2952413/examplePositionDirect.txt.zip)
rpc commands:
```bash
teo@teo-oliver:~$ yarp rpc /teo/rightArm/rpc:i 
>>set pos 3 0
Response: [ok]
>>set pos 3 0
Response: [ok]
>>set icmd cmds (pos pos pos pos pos pos pos)
Response: [ok]
>>get icmd cmds
Response: [is] [cmds] ([pos] [pos] [pos] [pos] [pos] [pos] [pos]) [tsta] 10 1552306524.43345 [ok]
>>set pos 3 0
Response: [ok]
>>set pos 0 5
Response: [ok]
>>set pos 0 5
Response: [ok]
>>set pos 1 -5
Response: [ok]
>>set pos 1 -5
Response: [ok]
>>set pos 1 0
Response: [ok]
>>set pos 3 0
Response: [ok]

```

Steps in the execution:
1. ExamplePositionDIrect is started (IPositionControl -> IPositionDirect )
2. We make a change to position mode by RPC of all the right arm
3. We check that the first time we try to move different joints by RPC, it does not react (joints 0, 1 and 3). We passed the position a second time. It reacts and we move them to 0.
4. We started the application again. It moves at 10 degrees correctly but then has made a sudden movement (I haven't had time to see it) and I've pressed the emergency button immediately.

PeterBowman, 2019-03-13:
> Change modes `IPositionControl -> IPositionDirect -> IPositionControl` and try to `positionMove()`: need to send this command twice, no response at first try.

See iPOS driver manual, 9.1.1. *Internal States*.

jgvictores, 2019-03-13:
> See iPOS driver manual, 9.1.1. _Internal States_.

Yes, that totally makes sense!

PeterBowman, 2019-03-15:
> Change modes and move to a certain position after each step, `IPositionControl -> IPositionDirect -> IPositionControl -> IPositionDirect`. We won't be able to start the PT mode since `ptPointCounter` is never set to zero:

Done at https://github.com/roboticslab-uc3m/yarp-devices/commit/6dae2643a8b033f52eaa8ac01fe9d3e24c041238, seems to work now.

rsantos88, 2019-03-15:
I'll upload the output of launchManipulation for this issue:
> Change modes IPositionControl -> IPositionDirect -> IPositionControl and try to positionMove(): need to send this command twice, no response at first try.

[launchManipulation-leftArm-frontal-wrist-1503.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/2970873/launchManipulation-leftArm-frontal-wrist-1503.txt.zip)

PeterBowman, 2019-03-19:
> [launchManipulation-leftArm-frontal-wrist-1503.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/2970873/launchManipulation-leftArm-frontal-wrist-1503.txt.zip)

I'm really baffled that only CAN IDs 21 (left arm) and 27 (head) are responding in the CBW2 state thread, almost no other IDs are shown there (just a few PDO-related lines). Perhaps the acceptance filters are misbehaving?

By the way, I noticed that the status word (6041h) transitions from 8637h to 9237h upon issuing a `positionMove()` that won't move the robot, in CAN ID 21 (shoulder). According to *8.1.4. Status word in profile position mode*, the *set-point acknowledge* bit is set (= "Trajectory generator will not accept a new set-point.") and the *target reached* one, reset (= "Target position not reached").

rsantos88, 2019-03-20:
The same test done now at 12:10 with left wrist (ID26)
[oneCanBusOneWrapper-posd-2003.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/2987844/oneCanBusOneWrapper-posd-2003.txt.zip)

* First: examplePositionDirect
* Then:
```bash
>>get encs
Response: [is] encs (20.003027) [tsta] 3 1553080045.375092 [ok]
>>get icmd cmds
Response: [is] [cmds] ([posd]) [tsta] 4 1553080049.250982 [ok]
>>set icmd cmod 0 pos
Response: [ok]
>>get icmd cmds
Response: [is] [cmds] ([pos]) [tsta] 5 1553080057.332439 [ok]
>>set pos 0 0  (not working)
Response: [ok]
>>set pos 0 0  (working)
Response: [ok]
```

rsantos88, 2019-03-20:
Control word test:
[oneCanBusOneWrapper-posd-2003-cw.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/2988004/oneCanBusOneWrapper-posd-2003-cw.txt.zip)

PeterBowman, 2019-03-20:
Remarks:

* Upon a faulty `positionMove`, the status word transitions from 96xxh to 86xxh (9h means that the *set-point acknowledge* bit is set = "Trajectory generator will **not** accept a new set-point.")
* Upon a successful `positionMove`, the status word transitions from 86xxh to 92xxh (2h means that the *target reached* bit is reset = "Target position **not** reached", I guess the motor is still rotating).
* The control word (6040h) transitions from 001fh to 000fh **after** the faulty `positionMove` is issued. Bit 8 means *Halt*:
  * 0 = "Execute positioning"
  * 1 = "Stop drive with profile acceleration"
* According to *8.1.3. Control word in profile position mode*, xfh means, most notably:
  * *New point*, 1 = "Assume target position (update the new motion parameters)"
  * *Change set immediately*, 1 = "Interrupt the actual positioning and start the next positioning. Valid only
for linear ramp profile" (see *8.1.2. Continuous motion profile*)

PeterBowman, 2019-03-20:
> Change modes `IPositionControl -> IPositionDirect -> IPositionControl` and try to `positionMove()`: need to send this command twice, no response at first try.

Done at https://github.com/roboticslab-uc3m/yarp-devices/commit/2616482c958fc17787884e137880ca2f38b98db4. The "10. Reset the set point." command has been replicated and now is sent prior to setting the target position in `positionMoveRaw()`. We noticed that *8.3.1. Absolute trapezoidal example* assumes the user waits for the motion to complete between "8. Start the profile." and "10. Reset the set point." (see "9. Wait movement to finish.") before sending the next target, therefore our first attempt was to move said reset-related lines to the top of this function. It did work in [pos] control mode, but prevented the trajectory from executing upon switching again to [posd]. As a workaround, we reset the set point **twice** in `positionMoveRaw()`.

PeterBowman, 2019-03-29:
> Change modes and move to a certain position after each step

Still broken. Previous (successful) tests did not cover a full arm, rather a single joint.

rsantos88, 2019-04-09:
> Still broken. Previous (successful) tests did not cover a full arm, rather a single joint. This time, on the second attempt (the first one is always successful, as noted above), we observe that the commanded joint rotates a bit in the beginning, then stops responding to new target points.

Detected the change that produces the error in the example. 
At a low level it would be necessary to see what is failing.
If we change [pos->positionMove(JOINT, 10)](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/examples/cpp/examplePositionDirect/examplePositionDirect.cpp#L66) for [pos->positionMove(v.data())](https://github.com/roboticslab-uc3m/yarp-devices/blob/position-direct-example/examples/cpp/examplePositionDirect/examplePositionDirect.cpp#L67), the example runs correctly.

rsantos88, 2019-04-09:
Test summary:
* In the example, we initialize all the joints in `POSITION` mode and we change also all the joints to `POSITION DIRECT`
* movement orders following **this order**: (1) Position -> (2) PositionDirect: 
**(1)** `positionMove(JOINT, 10)` **(2)** `setPosition(JOINT, encValue)` :heavy_check_mark:
**(1)** `positionMove(JOINT, 10)` **(2)** `setPosition(jointPosition.data())` :x:
**(1)** `positionMove(jointPosition.data())` **(2)** `setPosition(JOINT, encValue))` :heavy_check_mark:
**(1)** `positionMove(jointPosition.data())` **(2)** `setPositions(jointPosition.data())` :heavy_check_mark:

rsantos88, 2019-04-09:
Last test, prints current pending reads in CBW2:
[launchManipulation-0904.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/3059091/launchManipulation-0904.txt.zip)

PeterBowman, 2019-04-09:
We've learned that `ptBuffer` is the root of all evil. The device simply stops accepting new PT-mode target points since program execution is stuck at [this `ptBuffer.wait()` instruction](https://github.com/roboticslab-uc3m/yarp-devices/blob/a023145ed8a77cbeca0109c405855179654bf282/libraries/YarpPlugins/TechnosoftIpos/IPositionDirectRawImpl.cpp#L30), which in turn causes YARP to complain about a growing queue of (unprocessed) messages. We believe that this semaphore might not be properly released on the first run of *examplePositionDirect* ([relevant lines](https://github.com/roboticslab-uc3m/yarp-devices/blob/a023145ed8a77cbeca0109c405855179654bf282/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L255-L266)). Also, note that nothing prevents us from calling `post()` more than once per `wait()` call, that is, the internal counter of the semaphore may exceed the value `1`. In the log file previously attached, it's worth noting that the "pt buffer full" message is not symmetric to "pt buffer empty" (just 3 occurrences of the latter on the second run) and relates to IDs 24/25/26, mostly.

This is highly undesirable and leads to undefined behavior, to say the least. I would rethink the whole buffer full/empty handling mechanism.

jgvictores, 2019-04-10:
Attaching the oldest code I've found: [locomotion.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/3062847/locomotion.zip) (from [private repo](http://robots.uc3m.es/svn/RH2repo/Development/software/contrib/superseded/locomotion/
) which has some interesting stuff)

PeterBowman, 2019-04-10:
Regarding

https://github.com/roboticslab-uc3m/yarp-devices/blob/a023145ed8a77cbeca0109c405855179654bf282/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L255-L266

and *9.2.3. Object 2072h: Interpolated position mode status*:

> Remark: when a status bit changes from this object, an emergency message with the code
0xFF01 will be generated. This emergency message will have mapped object 2072h data onto
bytes 3 and 4.

PeterBowman, 2019-04-10:
> Attaching the oldest code I've found: [locomotion.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/3062847/locomotion.zip) (from [private repo](http://robots.uc3m.es/svn/RH2repo/Development/software/contrib/superseded/locomotion/) which has some interesting stuff)

This code set buffer-low signaling value to 4, buffer-empty was never checked against. There was an attempt to change the default buffer size (full) to 10792 (2A28h), which was ultimately commented out.

rsantos88, 2019-04-10:
Test with `oneCanBusOneWrapper`, prints the current changes of branch https://github.com/roboticslab-uc3m/yarp-devices/tree/verbose-posd with the position direct example.
File: [OneCanBusOneWrapper-1004.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/3063934/OneCanBusOneWrapper-1004.txt.zip)

PeterBowman, 2019-04-10:
> Shouldn't the "pt buffer empty" check look for value 80h, instead? That is, bit 15 set: "Buffer is empty – there is no point in the buffer."

Well, I can confirm the buffer is definitely not empty until the very end of the motion sequence. The buffer-low bit is set, too, in this case. After the last `setPosition()` call, and around 6-7 cycles later, a single buffer-low message is issued, then buffer-empty steps in. Also, on the second (failed) run, the semaphore gets stuck right after signaling a buffer-full state.

PeterBowman, 2019-04-10:
* https://www.technosoftmotion.com/ESM-um-html/index.html?tml_pt_mode.htm
* https://www.technosoftmotion.com/ESM-um-html/index.html?motion_module_pt.htm
* https://www.technosoftmotion.com/ESM-um-html/index.html?motion_module_linear_interpolation.htm

> Upon reception, each PT point is stored in a reception buffer. The reference generator empties the buffer as the PT points are executed. The drive/motor automatically sends warning messages when the buffer is full, low or empty.  The buffer full condition occurs when the number of PT points in the buffer is equal with the buffer size. The buffer low condition occurs when the number of PT points in the buffer is less or equal with a programmable value. The buffer empty condition occurs when the buffer is empty and the execution of the last PT point is over.

> The PT buffer size is programmable and if needed can be substantially increased. By default it is set to 7 PT points.

smcdiaz, 2019-04-10:
Good comment.  I don't remember if I talked about that with you or only with  @rsantos88 . Using these messages would be a good method for controlling movement and avoiding interruptions.

PeterBowman, 2019-04-11:
Ideas:

* Use an intermediate buffer on the server (device) side in order to decouple receive-from-yarp-network and send-to-ipos-drive operations. Alternatively, increase substantially the internal buffer size. Buffer state management is needed, anyway, so that the motion start signal is sent at the right time. This buffer would make sense in an offline trajectory scenario, that is, an external application might send lots of target points in large bursts, even hundreds or thousands at once, as long as YARP comms can handle that. In this manner, the latency of both YARP and CAN networks is entirely avoided. However, I would strive to avoid buffering altogether when processing online trajectories. Keep in mind that we feed this internal buffer with points and don't execute the movement until it's empty - that is, we are introducing a rather big delay between the instant the first target is meant to be executed and the actual start of the motion (buffer_size*T).

* Either enforce a PT interval (currently 50 milliseconds) that external apps should download and adjust interpolation input data to (online/offline), or process this data as it goes in said app and defer proper interpolation to the drive layer, i.e. let the TechnosoftIpos device choose the right PT interval. The former feels like reinventing the wheel and moves the burden of sending the exact points in the right time intervals to the client app. The latter forces our TechnosoftIpos to process input targets prior to forwarding them to the iPOS device via CAN, but avoids duplication (this task is accomplished in one place, external apps don't need to worry about the exact protocol).

* Perhaps other modes of operation may render more suitable for our needs than PT (*5.2.4. Object 6060h: Modes of Operation*):
  * Electronic Gearing Position Mode: one or more slave drives follow the position of a master drive. The master drive itself is another iPOS, therefore this mode is solely intended for replication of motion of one axis (we'd like to use the CAN master as a source of reference points for multiple axes).
  * Electronic Camming Position Mode: if I understand this correctly, it's quite similar to the previous mode, but we need to load the full camming profile (= trajectory) in memory.
  * External Reference Position Mode: perhaps?

PeterBowman, 2019-04-12:
> External Reference Position Mode: perhaps?

Tested today, we learned that its behavior closely resembles our IPositionDirect interface implementation in OpenRAVE: try to move as fast as possible to the demanded position. We did not manage to make the speed limit work.

PeterBowman, 2019-04-21:
Useful robotology PDFs regarding control mode implementation in the iCUB robot: [doc](https://github.com/roboticslab-uc3m/yarp-devices/files/3101452/ICub_Control_Modes_1_1.pdf) ([slides](https://github.com/roboticslab-uc3m/yarp-devices/files/3101451/Icub_control_modes_slides_1_1.pdf)). Source: http://wiki.icub.org/wiki/Control_Modes.

jgvictores, 2019-04-22:
> Useful robotology PDFs regarding control mode implementation in the iCUB robot: [doc](https://github.com/roboticslab-uc3m/yarp-devices/files/3101452/ICub_Control_Modes_1_1.pdf) [...]

@PeterBowman Thanks! Added to https://github.com/robotology/yarp/issues/1105

PeterBowman, 2019-04-27:
I reached out Technosoft support team and learned that:

* Default buffer length in PT and PVT modes is 9 and 7 targets, respectively.
* Maximum buffer length in PT and PVT modes is 285 and 222 targets, respectively (2000 words).
* A PT target takes 7 words of space in the buffer, whereas a PVT target takes 9 words.
* PT/PVT points are stored in a RAM memory location shared by this mode, the logger and TML programs; care should be taken that there are no overlaps happening.
* The logger is only used for tuning / troubleshooting purposes in the EasyMotionStudio application.
* The logger is automatically started once by ESM when the RUN button is pressed.
* Extending the PT\PVT buffer size to the max. value implies no space for the logger.
* Memory usage can be accessed at *Application General Information* > *Memory Settings* in EMS.
* RAM defaults: 0x40 = 64words from 9800h to 983Fh for buffer and 0x790 = 1936words from 9840h to 9FCFh for logger.
* Although 63 words are actually used, it is set to 64 since the logger can only start from an address value which is a multiple of 4.
* The recommended way to increase the PT/PVT buffer size is to modify the "PVTsize" parameter in the "setup.cfg" file from the application folder.
* PT/PVT buffer lenght can be set via object 2073h. It is not persistent - this object needs to be init every time we power-on/reset the drive.
* Upon buffer empty when in PT mode, the drive will simply remain in PT mode, hold position and patiently wait for new points to execute.
* In PVT mode, the drive will remain in PVT mode and wait for new points only if the last point is a zero velocity point (this corresponds to a trajectory finished situation).
* If the last executed PVT point didn't had a zero speed command, then this is considered as an error situation (a normal trajectory can't finish with a speed different than zero) and the drive will change the PVT mode to quickstop mode where the motor speed is decelerated to zero based on the value of the CDEC variable. The PVT mode needs to be set again before the drive will accept and execute new PVT points.

PeterBowman, 2019-04-28:
The *Technosoft iPOS CANopen Programming User Manual* (ref. [P091.063.iPOS.STO.UM.0117](https://github.com/roboticslab-uc3m/yarp-devices/files/3124933/P091.063.CANopen.iPOS.UM.634.pdf), 2017) adds a **Cyclic Synchronous Position mode (CSP)** in section 10:

> With this mode, the trajectory generator is located in the control device, not in the drive device. In cyclic synchronous manner, it provides a target position to the drive device, which performs position control, velocity control and torque control. Measured by sensors, the drive provides actual values for position, velocity and torque to the control device.

Section 9.2.1, *Object 60C0h: Interpolation sub mode select* (in *Interpolated Position Mode*), states that PT and PVT submodes should be regarded as legacy stuff. The preferred submode is introduced as "linear interpolation":

> Linear interpolation as described in the CiA 402 standard (when object 208Eh bit8=1); This mode is almost identical with Cyclic Synchronous Position mode, only that it receives its position data into 60C1h sub-index 01 instead of object 607Ah. No interpolation point buffer will be used.

From *About This Manual*:

> The iPOS drives are conforming to CiA 301 v4.2 application layer and communication profile, CiA WD 305 v.2.2.131 Layer Setting Services and to CiA (DSP) 402 v4.0 device profile for drives and motion control, now included in IEC 61800-7-1 Annex A, IEC 61800-7-201 and IEC 61800-7-301 standards.

[P091.063.iPOS.UM.0615](https://github.com/roboticslab-uc3m/yarp-devices/files/3124947/P091.063.CANopen.iPOS.UM.pdf) (2015) states that:

> The iPOS drives are confirming to CiA 301 v4.2 application layer and communication profile, CiA WD 305 v.2.2.131 Layer Setting Services and to CiA DSP 402 v3.0 device profile for drives and motion control (...)

So, no CSP mode in CiA DSP 402 v3.0. We'd probably need to perform a firmware update on our drives in order to use this mode.

Interestingly, from section 10.2.1, *Object 60C2h: Interpolation time period*:

> The Interpolation time period indicates the configured interpolation cycle time. Its value must be set with the time value of the CANopen master communication cycle time and sync time in order for the Cyclic Synchronous Position mode to work properly.
>
> Remark: due to the limitations of the CAN network, it is recommended that the interpolation time period should not be set lower than 4 ms.

PeterBowman, 2019-04-28:
Found by @jgvictores at https://www.technosoftmotion.com/en/features (beware: 2006):

> Position and speed control with 100 ms update rate - for very high dynamic applications a new control strategy has been added. This performs the position and speed control with an update rate of 100 ms, and does linear interpolation between the reference points provided by the reference generator at each 1ms.

PeterBowman, 2019-04-28:
As I understand it, the external reference position mode is *stupid* in that is doesn't perform any control (position, velocity, current) nor interpolation on the drive side. Position profile mode places the burden of control and trajectory generation on the drive. There is also a need for interpolation (i.e. trajectory generation) on legacy PT/PVT modes, plus an input buffer is supported. Control is performed by the drive in CSP mode, but the application (CAN master) is now responsible for the trajectory/interpolation step.

PeterBowman, 2019-04-28:
Current plan:

* Repurpose the IPositionDirect implementation, command the iPOS drives in the external reference position mode, take care of setting the speed limitation.

* Enable PVT interpolation via [IRemoteVariables](https://github.com/robotology/yarp/blob/32ba9f4d4bf7ec8d70fc586a148b491d4e51dab9/src/libYARP_dev/include/yarp/dev/IRemoteVariables.h):
  * On the YARP side, use [`VOCAB_CM_MIXED`](https://github.com/robotology/yarp/blob/32ba9f4d4bf7ec8d70fc586a148b491d4e51dab9/src/libYARP_dev/include/yarp/dev/IControlMode.h#L138) a.k.a. mixed mode ([iCub reference](https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-485269958)).
  * Send a list (bottle) of PVT targets via IRemoteVariables.
  * Send a start signal along with this bottle. Mind synchronization across joints (first fill the buffer, then start motion).
  * Progressively fill the internal iPOS buffer using an intermediate buffer and a periodic thread.
  * Idea: increase the PVT and ICanBus buffer sizes.
  * If only PT information is available, estimate instantaneous speed between consecutive targets.
  * Set zero speed on the last target, but allow more points to be appended via `IRemoteVariables::setRemoteVariable`.
  * Enable `IPositionControl::checkMotionDone` and `::stop` (https://github.com/roboticslab-uc3m/yarp-devices/issues/120) in this mode.

Side quests:

* Implement offline trajectories in [YarpOpenraveControlboard](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/5b0c5640269cdf1e39e07310cccbfe6e770ee2ab/libraries/YarpPlugins/YarpOpenraveControlboard).
* Same in [`BasicCartesianControl::movl`](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/530561b45c238a6f18cffa30d20d90bb25ac587c/libraries/YarpPlugins/BasicCartesianControl) (and others).

jgvictores, 2019-04-28:
> * Implement offline trajectories in [YarpOpenraveControlboard](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/5b0c5640269cdf1e39e07310cccbfe6e770ee2ab/libraries/YarpPlugins/YarpOpenraveControlboard).

Yes, I guess we may end up with very similar code on that side. Cannot find good alternative.

PeterBowman, 2019-04-30:
> We did not manage to make the speed limit work.

Fixed. YARP's position-direct mode is now implemented with the iPOS-specific external reference position mode, was tested successfully today. WIP in the nuke-posd branch, commit https://github.com/roboticslab-uc3m/yarp-devices/commit/622dbf0de971f06f9d154ebe3732bb2ee264cdcb might be merged into develop soon.

PeterBowman, 2019-04-30:
https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-487386797

> Repurpose the IPositionDirect implementation, command the iPOS drives in the external reference position mode, take care of setting the speed limitation.

Done, awaiting some final tests.

> Enable PVT interpolation via [IRemoteVariables](https://github.com/robotology/yarp/blob/32ba9f4d4bf7ec8d70fc586a148b491d4e51dab9/src/libYARP_dev/include/yarp/dev/IRemoteVariables.h):

Split into https://github.com/roboticslab-uc3m/yarp-devices/issues/208. This issue only focuses on posd mode, I've updated the description accordingly.

PeterBowman, 2019-05-05:
> Done, awaiting some final tests.

For the sake of caution, no tests shall be performed until - and including - next Thursday, May 9th (@rsantos88). A video recording will be held on that date.

After that, I'd like to test extensively our new online trajectory execution with some kind of joystick (see https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173). Idea: dynamically adjust the maximum speed limitation on the driver side (could be done by the client, but I'd rather do this over the CAN network rather than via Ethernet).

PeterBowman, 2019-05-17:
> After that, I'd like to test extensively our new online trajectory execution with some kind of joystick

Ran into this: "No response to user commands after the application is relaunched." (https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173#issuecomment-493443729).

> Idea: dynamically adjust the maximum speed limitation on the driver side (could be done by the client, but I'd rather do this over the CAN network rather than via Ethernet).

The CAN master would need to take account of the frequency at which commands are sent. This statement does not cope well with my reasoning about velocity checks at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173#issuecomment-493191279, so I'd need to take another look at it.

PeterBowman, 2019-05-17:
New posd implementation merged into develop at https://github.com/roboticslab-uc3m/yarp-devices/commit/36f8fad94ea34c5046491540d3361d7e84e73de8, not fully ready yet (see previous comment).

PeterBowman, 2019-05-18:
> Found by @jgvictores at https://www.technosoftmotion.com/en/features (beware: 2006):

According to https://github.com/roboticslab-uc3m/yarp-devices/issues/210#issuecomment-493665173, the sampling period for fast loop and slow loop control is 0.1 and 1 ms, respectively.

PeterBowman, 2019-05-18:
> Idea: dynamically adjust the maximum speed limitation on the driver side (could be done by the client, but I'd rather do this over the CAN network rather than via Ethernet).

Also, blocked by https://github.com/roboticslab-uc3m/yarp-devices/issues/188.

PeterBowman, 2019-05-22:
Even if the reference speed is updated while in external reference position mode, its new values are ignored by the drive upon trying to reach the next targets. Also, this mode is sometimes unresponsive upon pos->posd->pos->posd, not so often when running the examplePositionDirect app.

Considering to switch back to PT mode for online trajectories, leaving PVT aside for offline stuff.

PeterBowman, 2019-05-23:
> Considering to switch back to PT mode for online trajectories, leaving PVT aside for offline stuff.

Implemented and tested. I'm working on this along with https://github.com/roboticslab-uc3m/yarp-devices/issues/208 since this solution uses the `IRemoteVariables` YARP interface to pass a `ptModeMs` parameter over the network: `<0` means "not in position direct mode" (default), `0` means "use external reference position mode", `>0` means "use PT interpolation mode with this period (in milliseconds)".

I noticed that device routing in the ControlBoardWrapper layer is kinda messy regarding how IRemoteVariables methods are interpreting incoming/outgoing bottles, see [source code](https://github.com/robotology/yarp/blob/86cfea57195a756fd08df5bc73757ad87fbdffe8/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L2848-L2911). For instance, we'd need to build a two-element bottle if we ever want to send a remote variable to the head part, one element per CanBusControlboard instance. This won't affect both arms and legs since there is a one-to-one mapping (controlboardwrapper2-to-CanBusControlboard), but we are still forced to wrap any data sent in an additional bottle (e.g. `(50)` instead of just `50` for `ptModeMs`).

PeterBowman, 2019-05-25:
Recent tests showed that online trajectories performed in the new position direct mode, (re)implemented as PT interpolation, suddenly stop roughly at the 150th-200th commanded target. The drives had turned unresponsive after that point: no lines received in the CAN read stream (`interpretMessage`) despite reporting apparently successful status queries. A close inspection in EasyMotion did not reveal much - I could only learn that all four position trigger bits in object 1002h (Manufacturer Status Register) had been toggled to zero. This behavior was observed in all drives accounting for the right arm (firmware F508H) except id 19 (firmware F508I).

Everything worked like a charm after I commented out the bits that set the PT buffer length to its maximum value as detailed in https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-487279910. This buffer has never been modified in that way before, even if we thought it was modified by us to store 15 points by the following lines:

https://github.com/roboticslab-uc3m/yarp-devices/blob/e26db48e8a0d6582df8ca18583b55f8b3bfba898/libraries/YarpPlugins/TechnosoftIpos/IControlModeRawImpl.cpp#L211-L215

This has been fixed in a recent revision of the iPOS CAN user manual: should have read object 2073h (Interpolated position buffer length) instead of 2074h (Interpolated position buffer configuration).

rsantos88, 2019-05-28:
Reminder:
When I do `pos->posd->pos->posd`, it doesn't change to PT mode and the launchManipulation shows:
```bash
[debug] IRemoteVariablesImpl.cpp:50 setRemoteVariable(): ptModeMs
[debug] IRemoteVariablesRawImpl.cpp:37 setRemoteVariableRaw(): ptModeMs
[warning] IRemoteVariablesRawImpl.cpp:45 setRemoteVariableRaw(): Illegal state, unable to change variable when currently used.
[warning] IRemoteVariablesImpl.cpp:58 setRemoteVariable(): Unable to set remote variable on node 0.
```

PeterBowman, 2019-05-28:
Could you upload the latest code? I don't see the changes we had to introduce in [this method](https://github.com/roboticslab-uc3m/teo-bimanipulation/blob/cc17c98e84fab6d42538a04a263bf794d899cb00/programs/BalanceTray/BalanceTray.cpp#L453-L478). There is a possibility the CAN network has nothing to do here, perhaps a state variable is not being correctly cleared on the iPOS-YARP device side, or some tweaks need to be made in your app.

rsantos88, 2019-05-28:
> Could you upload the latest code? I don't see the changes we had to introduce in [this [...]

https://github.com/roboticslab-uc3m/teo-bimanipulation/commit/86b384a8e37efa3c64e01b8e5e288c7613b2b069

PeterBowman, 2019-05-29:
Please run your applicacion twice, make sure this error happens on the second run, and paste here the complete logs of launchManipulation.

rsantos88, 2019-05-30:
Here you have the output information of launchManipulation:
[launchManipulation-pvt-test-3005.txt.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/3236340/launchManipulation-pvt-test-3005.txt.zip)

And the output of my app:
```bash
$ ./balanceTray 
||| clearing context
||| adding context [balanceTray]
||| configuring
Run "./balanceTray --help" for options.
./balanceTray checking for yarp network... [ok]
inicia el configure
--------------------------------------------------------------
Mode KEYBOARD activated [OK]
yarp: Port /balanceTray/teo/rightArm/rpc:o active at tcp://2.2.2.101:10053/
yarp: Port /balanceTray/teo/rightArm/command:o active at tcp://2.2.2.101:10054/
yarp: Port /balanceTray/teo/rightArm/stateExt:i active at tcp://2.2.2.101:10055/
yarp: Sending output from /balanceTray/teo/rightArm/rpc:o to /teo/rightArm/rpc:i using tcp
yarp: Sending output from /balanceTray/teo/rightArm/command:o to /teo/rightArm/command:i using udp
yarp: Receiving input from /teo/rightArm/stateExt:o to /balanceTray/teo/rightArm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[info] BalanceTray.cpp:78 configure(): Acquired leftArmIEncoders interface
[info] BalanceTray.cpp:87 configure(): Acquired rightArmIControlMode interface
[info] BalanceTray.cpp:94 configure(): Acquired rightArmIPositionControl interface
[info] BalanceTray.cpp:101 configure(): Acquired rightArmIPositionDirect interface
[info] BalanceTray.cpp:108 configure(): Acquired rightArmIRemoteVariables interface
yarp: Port /balanceTray/teo/leftArm/rpc:o active at tcp://2.2.2.101:10056/
yarp: Port /balanceTray/teo/leftArm/command:o active at tcp://2.2.2.101:10057/
yarp: Port /balanceTray/teo/leftArm/stateExt:i active at tcp://2.2.2.101:10058/
yarp: Sending output from /balanceTray/teo/leftArm/rpc:o to /teo/leftArm/rpc:i using tcp
yarp: Sending output from /balanceTray/teo/leftArm/command:o to /teo/leftArm/command:i using udp
yarp: Receiving input from /teo/leftArm/stateExt:o to /balanceTray/teo/leftArm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[info] BalanceTray.cpp:130 configure(): Acquired leftArmIEncoders interface
[info] BalanceTray.cpp:140 configure(): Acquired leftArmIControlMode interface
[info] BalanceTray.cpp:147 configure(): Acquired leftArmIPositionControl interface
[info] BalanceTray.cpp:154 configure(): Acquired leftArmIPositionDirect interface
[info] BalanceTray.cpp:161 configure(): Acquired leftArmIRemoteVariables interface
---- Joint limits of right-arm ----
Joint 0 limits: [-98.100000,106.000000]
Joint 1 limits: [-75.500000,22.400000]
Joint 2 limits: [-80.100000,57.000000]
Joint 3 limits: [-99.600000,98.400000]
Joint 4 limits: [-80.400000,99.600000]
Joint 5 limits: [-115.100000,44.700000]
Joint 6 limits: [-1200.000000,1200.000000]
[debug] DeviceDriverImpl.cpp:134 open(): config: (H0 (0 1 0 0 0 0 1 -0.3469 1 0 0 0.4982 0 0 0 1)) (HN (0 0 -1 -0.0975 -1 0 0 0 0 1 0 0 0 0 0 1)) (device KdlSolver) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_1 (offset -90.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_2 (offset -90.0) (D -0.32901) (A 0.0) (alpha -90.0) (mass 1.750625) (cog 0 0 0) (inertia 0 0 0)) (link_3 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_4 (offset 0.0) (D -0.215) (A 0.0) (alpha -90.0) (mass 2.396) (cog 0 0 0) (inertia 0 0 0)) (link_5 (offset -90.0) (D 0.0) (A -0.09) (alpha 0.0) (mass 0.3) (cog 0 0 0) (inertia 0 0 0)) (maxs (106.0 22.4 57.0 98.4 99.6 44.7 1200.0)) (mins (-98.1 -75.5 -80.1 -99.6 -80.4 -115.1 -1200.0)) (numLinks 6).
[info] DeviceDriverImpl.cpp:138 open(): kinematics: none.ini [none.ini]
||| did not find none.ini
yarp: cannot read from 
[debug] DeviceDriverImpl.cpp:149 open(): fullConfig: (H0 (0 1 0 0 0 0 1 -0.3469 1 0 0 0.4982 0 0 0 1)) (HN (0 0 -1 -0.0975 -1 0 0 0 0 1 0 0 0 0 0 1)) (device KdlSolver) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_1 (offset -90.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_2 (offset -90.0) (D -0.32901) (A 0.0) (alpha -90.0) (mass 1.750625) (cog 0 0 0) (inertia 0 0 0)) (link_3 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_4 (offset 0.0) (D -0.215) (A 0.0) (alpha -90.0) (mass 2.396) (cog 0 0 0) (inertia 0 0 0)) (link_5 (offset -90.0) (D 0.0) (A -0.09) (alpha 0.0) (mass 0.3) (cog 0 0 0) (inertia 0 0 0)) (maxs (106.0 22.4 57.0 98.4 99.6 44.7 1200.0)) (mins (-98.1 -75.5 -80.1 -99.6 -80.4 -115.1 -1200.0)) (numLinks 6).
[info] DeviceDriverImpl.cpp:153 open(): numLinks: 6 [1]
[info] DeviceDriverImpl.cpp:165 open(): gravity: 0.0 0.0 -9.81 [0.0 0.0 -9.81]
[info] DeviceDriverImpl.cpp:181 open(): H0:
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	-0.346900
 1.000000	 0.000000	 0.000000	 0.498200
 0.000000	 0.000000	 0.000000	 1.000000
[ 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000]
[info] DeviceDriverImpl.cpp:208 open(): Added: link_0 (offset 0.000000) (D 0.000000) (A 0.000000) (alpha -90.000000) (mass 0.000000) (cog 0.000000 0.000000 0.000000) (inertia 0.000000 0.000000 0.000000)
[... cut ...]
R-arm pose: [0.354048 -0.001619 0.260271 -0.377632 0.619060 -0.688592 0.430233 ]
L-arm pose: [0.355017 0.001193 0.263784 -0.235751 0.547195 0.803119 0.458293 ]
Put the tray and press a Key...

[info] BalanceTray.cpp:479 configArmsToPositionDirect(): Configuring drivers to Position Direct...
[error] BalanceTray.cpp:487 configArmsToPositionDirect(): Problems setting ptModeMs remote variable of: right-arm
[error] BalanceTray.cpp:309 configure(): Configuring drivers to Position Direct
[INFO]RFModule failed to open.

```

PeterBowman, 2019-05-30:
Identified at https://github.com/roboticslab-uc3m/yarp-devices/issues/211#issuecomment-497280755. Suggested workaround: edit launchManipulation.ini, remove head joints.

PeterBowman, 2019-07-20:
Just tried another implementation which does not need to care about synchronizing send rates on remote client and CAN master. Each TechnosoftIpos node starts a separate thread given a fixed period (50 ms). By the way, I'm giving another try to PVT submode, therefore the internal buffer (shrinked to hold a single target) must never deplete. ASBJ (as suggested by @jgvictores), the thread mimicks a LIFO stack in that only the last received target is forwarded to the buffer via CAN. In other words, high-frequency (<50 ms) trajectories get downsampled, whereas the last commanded target is placed in the buffer if no new point was received within 50 ms.

Checklist:

- Make sure PVT is the right choice. PT might perform better and safer (it's OK to reach buffer empty state).

- iPOS drives will generate an EMCY message each time a point is received because of the single-element-size buffer. I guess this point is never replaced by successively arriving targets unless it's consumed by the PVT interpolator, thus making room for more points. Let's check that. Also make sure that points are actually discarded in that case.

- PVT target processing sequence behaves like so: point is stored, therefore buffer is bull; point is consumed and buffer low state is signalized; a new point arrives and fills the buffer, thus it never depletes. Current implementation on posd mode change starts PVT thread prior to starting drive's PVT mode. Inverting this sequence turns the posd commands unresponsive, need to check why:

  https://github.com/roboticslab-uc3m/yarp-devices/blob/08d82af78fd643dae61411668934d93bdf42a268/libraries/YarpPlugins/TechnosoftIpos/IControlModeRawImpl.cpp#L199-L239

- I observed a rather poor behavior while commanding the robot in an online manner (spnav). Reminds me of external reference position mode, which was unable to cope with distinct per-joint trajectory durations (same speed on all joints). This time, it's a bit worse since joints appear to move in a non-synchronized manner. My hunch is that PVT threads must send points simultaneously. Preferably, a single thread started on the CanBusControlboatd level would take care of this.

PeterBowman, 2019-07-22:
Follow-ups to my previous post (WIP):

> Forgot to run _examplePositionDirect_ on different rates. Test <50 ms and >50 ms.

OK with smaller rates, loses smoothness at higher values (as expected).

PeterBowman, 2019-07-25:
> So, no CSP mode in CiA DSP 402 v3.0. We'd probably need to perform a firmware update on our drives in order to use this mode.

Updated ID 9 (left leg) to F508M (latest one in F508 series) and checked supported drive operation modes thanks to https://github.com/roboticslab-uc3m/yarp-devices/commit/1aef421ba7006311644fa3960c2fa1e6b3d2f5b2. Still no CSP mode available, although I had to check other manufacturer's description pertaining to this CAN object since the iPOS user manual does not cover CSP at all in 6502h.

```
[info] ICanBusSharerImpl.cpp:1084 interpretMessage(): Got "Supported drive modes" from driver. 43 2 65 0 65 0 1f 0. canId(9) via(580).
[info] ICanBusSharerImpl.cpp:1088 interpretMessage(): 	*profiled position (pp)
[info] ICanBusSharerImpl.cpp:1096 interpretMessage(): 	*profiled velocity (pv)
[info] ICanBusSharerImpl.cpp:1104 interpretMessage(): 	*homing (hm)
[info] ICanBusSharerImpl.cpp:1108 interpretMessage(): 	*interpolated position (ip)
[info] ICanBusSharerImpl.cpp:1124 interpretMessage(): 	*electronic camming position (manufacturer specific)
[info] ICanBusSharerImpl.cpp:1128 interpretMessage(): 	*electronic gearing position (manufacturer specific)
[info] ICanBusSharerImpl.cpp:1132 interpretMessage(): 	*external reference position (manufacturer specific)
[info] ICanBusSharerImpl.cpp:1136 interpretMessage(): 	*external reference speed (manufacturer specific)
[info] ICanBusSharerImpl.cpp:1140 interpretMessage(): 	*external reference torque (manufacturer specific)
```

PeterBowman, 2019-07-26:
Change mode, set first PT point, don't send start signal:

```
[debug] LinearInterpolationBuffer.cpp:165 configureMessage(): Sending p -0.009668 t 50 (ic 0).
[info] ICanBusSharerImpl.cpp:433 interpretMessage(): Interpolated position mode status. canId: 20.
[info] ICanBusSharerImpl.cpp:437 interpretMessage(): 	* buffer is not empty.
[info] ICanBusSharerImpl.cpp:458 interpretMessage(): 	* buffer is not low.
[info] ICanBusSharerImpl.cpp:471 interpretMessage(): 	* buffer is full.
[info] ICanBusSharerImpl.cpp:476 interpretMessage(): 	* no integrity counter error.
```

Once (iff) start signal is sent, the buffer is being processed by the linear interpolator:

```
[success] TechnosoftIpos.cpp:97 sendLinearInterpolationStart(): Sent "startPT". 1f 0. canId(20) via(200).
[...]
[info] ICanBusSharerImpl.cpp:433 interpretMessage(): Interpolated position mode status. canId: 20.
[info] ICanBusSharerImpl.cpp:437 interpretMessage(): 	* buffer is not empty.
[info] ICanBusSharerImpl.cpp:462 interpretMessage(): 	* buffer is low.
[info] ICanBusSharerImpl.cpp:467 interpretMessage(): 	* buffer is not full.
[info] ICanBusSharerImpl.cpp:476 interpretMessage(): 	* no integrity counter error.
```

Shortly thereafter:

```
[info] ICanBusSharerImpl.cpp:433 interpretMessage(): Interpolated position mode status. canId: 20.
[info] ICanBusSharerImpl.cpp:441 interpretMessage(): 	* buffer is empty.
[info] ICanBusSharerImpl.cpp:462 interpretMessage(): 	* buffer is low.
[info] ICanBusSharerImpl.cpp:467 interpretMessage(): 	* buffer is not full.
[info] ICanBusSharerImpl.cpp:476 interpretMessage(): 	* no integrity counter error.
```

Conclusion: buffer empty state is triggered when there are no more points in the buffer AND the last point has been already processed. If the buffer low signal is configured for buffer size zero, it is sent to the CAN master a bit earlier, that is, when that last point is being consumed (thus producing motion).

PeterBowman, 2019-07-29:
The position direct control mode has been rewritten (again), this time it should be final. See https://github.com/roboticslab-uc3m/yarp-devices/pull/220 for implementation details. To sum up, we tried four different setups:

#### old PT mode

* was filling the internal iPOS buffer with seven points, using T=50 milliseconds
* unsuitable for online trajectories because of the massive delay (`7*50=350` ms)
* suffered from several bugs, see intro comment at this very issue
* https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-482174998

#### external reference position mode

* aimed for execution of online trajectories (no buffer involved)
* kinda jerky since zero-velocity states are almost always reached between successive points
* joint coordination cannot be achieved, every motor reaches its final target at a different instant
* we are unable to change the reference speed unless leaving and entering this mode again
* https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-487386797

#### PT mode with synchronized client-master period

* in order to avoid those zero-velocity intervals, command send periods must be tightly coupled with the internal PT period (the "T" time)
* the `yarp::dev::IRemoteVariables` interface was implemented so that clients could set the T parameter in an online fashion
* this interface is buggy (https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-495322968) and unwieldy (requires additional code on the client side, see https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173)
* https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-495322968

##### PT/PVT modes with downsampling

Learn more at https://github.com/roboticslab-uc3m/yarp-devices/pull/220.

closed 2019-07-29 by PeterBowman

PeterBowman, 2019-07-30:
Regarding CSP mode... Just discovered object 208Eh, *Auxiliary Settings Register*:

* Bit 8 = 0: Set interpolation mode compatible with PT and PVT (legacy)
* Bit 8 = 1: Set interpolation mode (when 6060=7) as described in the CiA402 standard

Default is 1, but this object does not exist in earlier versions of the iPOS user manual.