roboticslab-uc3m/yarp-devices #203: Change Jr3 Filter value

jmgarciah, 2019-02-14:
When I have to do a test or record a video, always I have to change the value of the filters for the FT sensors (both ankles and wrists).

At this moment, the default value for the four filters is 0 ` ret = ioctl(fd,IOCTL0_JR3_FILTER0,&fm0);`

Can we replace in [IAnalogSensorImpl.cpp](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/libraries/YarpPlugins/Jr3/IAnalogSensorImpl.cpp) this filter 0 with the filter 2 ?? The filter 2 has a higher frequency, so it can filter the noise of the sensor and measure the external perturbation quickly. It is not as restrictive as the filter 6.

jgvictores, 2019-02-15:
Seeing that there are no methods to support this at [IAnalogSensor](http://www.yarp.it/classyarp_1_1dev_1_1IAnalogSensor.html), I'd start by including this as a parameter passed to the device parsed within the `open()` method: https://github.com/roboticslab-uc3m/yarp-devices/blob/ff690808731d3c75cd598f9a7179030b2df768f5/libraries/YarpPlugins/Jr3/DeviceDriverImpl.cpp#L9

Something like `--filter 0` and `--filter 2` to map to these values?

jmgarciah, 2019-02-15:
It is a great idea if we can use different filters by including this parameter to the device

PeterBowman, 2019-05-20:
Kinda unrelated, but I noticed that Clang complains about these lines:

https://github.com/roboticslab-uc3m/yarp-devices/blob/a538b382ddec2649c7c1f1ea9400be867ac49170/libraries/YarpPlugins/Jr3/DeviceDriverImpl.cpp#L24-L26

Each `5.5` value is narrowed down to `5` in order to fit into an integer-type variable.

PeterBowman, 2019-07-29:
> Each `5.5` value is narrowed down to `5` in order to fit into an integer-type variable.

Related to https://github.com/roboticslab-uc3m/yarp-devices/issues/184?

Also, I guess this issue is meant to make https://github.com/roboticslab-uc3m/yarp-devices/commit/fb1c1f64a234ab4a62f36da26b206ef854cdb251 subject to parameterization.

PeterBowman, 2019-07-29:
> I'd start by including this as a parameter passed to the device parsed within the `open()` method

@jgvictores, @jmgarciah would https://github.com/roboticslab-uc3m/yarp-devices/commit/8ba58cec870fcd19495d9a858dbf68dabc6f2a28 match the expected behavior?

PeterBowman, 2019-08-03:
Merged at https://github.com/roboticslab-uc3m/yarp-devices/commit/90a0aca3e6fb055ab9f375c69e4979848674d117.

closed 2019-08-03 by PeterBowman