roboticslab-uc3m/yarp-devices #211: Simplify inheritance chain in raw subdevices

PeterBowman, 2019-05-04:
These devices are wrapped by CanBusControlboard (CBC):

* [CuiAbsolute](https://github.com/roboticslab-uc3m/yarp-devices/tree/d747fa641e798d3a606e6aecf25f04bcbdff96d4/libraries/YarpPlugins/CuiAbsolute)
* [FakeJoint](https://github.com/roboticslab-uc3m/yarp-devices/tree/d747fa641e798d3a606e6aecf25f04bcbdff96d4/libraries/YarpPlugins/FakeJoint)
* [LacqueyFetch](https://github.com/roboticslab-uc3m/yarp-devices/tree/d747fa641e798d3a606e6aecf25f04bcbdff96d4/libraries/YarpPlugins/LacqueyFetch)
* [TechnosoftIpos](https://github.com/roboticslab-uc3m/yarp-devices/tree/d747fa641e798d3a606e6aecf25f04bcbdff96d4/libraries/YarpPlugins/TechnosoftIpos)
* [TextilesHand](https://github.com/roboticslab-uc3m/yarp-devices/tree/d747fa641e798d3a606e6aecf25f04bcbdff96d4/libraries/YarpPlugins/TextilesHand)

It just feels wrong to have CuiAbsolute implement several YARP motor interfaces:

https://github.com/roboticslab-uc3m/yarp-devices/blob/d747fa641e798d3a606e6aecf25f04bcbdff96d4/libraries/YarpPlugins/CuiAbsolute/CuiAbsolute.hpp#L43-L45

Ideas:

* CanBusControlboard should not crash if some of these subdevices don't inherit an unrelated interface.
* Introduce roles, e.g. motor devices (TechnosoftIpos and FakeJoint should derive from IPositionControl, IControlMode and so on), helpers (CuiAbsolute), end-effectors (LacqueyFetch, TextilesHand)...
  * Mind the Dextra hand device: https://github.com/roboticslab-uc3m/yarp-devices/issues/176.
  * Perhaps a new omnibus interface could come in handy, e.g. make TechnosoftIpos derive from IMotorDevice, which in turn derives from standard YARP interfaces and ICanBusSharer, but doesn't introduce any new methods by itself.
  * Split CuiAbsolute configuration into a new group within .ini files? Useless interfaces aside, we also don't need to pass a default max vel/acc value nor several others.
* What is IInteractionMode and why do we need it? Drop unused interfaces.

PeterBowman, 2019-05-22:
I might have discovered a massive pitfall in the way CanBusControlboard forwards calls to raw subdevices, see https://github.com/roboticslab-uc3m/yarp-devices/issues/214#issuecomment-494901096. It is imperative to be aware of which joints are actually controlled via YARP ports (look for `controlledJoints` at [ControlBoardWrapper.cpp](https://github.com/robotology/yarp/blob/86cfea57195a756fd08df5bc73757ad87fbdffe8/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp)). The YARP wrapper clips our list of nodes (iPOS+encoders+grippers+...) to the number specified in the `joints` property, see [launchManipulation.ini](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/f1c5cb41c9993fea3611342036f7e0dfd6ffc192/share/launchManipulation/conf/launchManipulation.ini#L37). Instead of iterating the full vector of `nodes` like in [here](https://github.com/roboticslab-uc3m/yarp-devices/blob/e26db48e8a0d6582df8ca18583b55f8b3bfba898/libraries/YarpPlugins/CanBusControlboard/IPositionControlImpl.cpp#L35-L38), CanBusControlboard should pick the correct ids, no more and no less than required.

PeterBowman, 2019-05-22:
Investigate:

* [`ImplementPositionControl::positionMove`](https://github.com/robotology/yarp/blob/86cfea57195a756fd08df5bc73757ad87fbdffe8/src/libYARP_dev/src/ImplementPositionControl.cpp#L95-L103)
* [ControlBoardHelper.hpp](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/libYARP_dev/include/yarp/dev/ControlBoardHelper.h#L167)
* [ControlBoardHelper.cpp](https://github.com/robotology/yarp/blob/86cfea57195a756fd08df5bc73757ad87fbdffe8/src/libYARP_dev/src/ControlBoardHelper.cpp)

PeterBowman, 2019-05-23:
Current workaround (unstable branch): https://github.com/roboticslab-uc3m/yarp-devices/commit/0c64a1ddb03dd5d1a53be478211bbabb7e631f6d, see `motorIds` class member.

PeterBowman, 2019-05-30:
> Current workaround (unstable branch): [0c64a1d](https://github.com/roboticslab-uc3m/yarp-devices/commit/0c64a1ddb03dd5d1a53be478211bbabb7e631f6d), see `motorIds` class member.

Every time we `setRemoteVariable`, all **seven** iPOS nodes on that CAN bus are being talked to (6 on the arm + 1 on the head). Then, we set control modes on the 6 arm joints only.

PeterBowman, 2019-06-03:
> I might have discovered a massive pitfall in the way CanBusControlboard forwards calls to raw subdevices

Another scenario regarding `IPositionControl::checkMotionDone` was described at https://github.com/roboticslab-uc3m/yarp-devices/issues/214#issuecomment-494901096.

Also, keep in mind that the vector of `nodes` includes non-iPOS devices, e.g. encoders, grabbers, etc.

PeterBowman, 2019-07-07:
I'd fancy a .ini file structure that doesn't enforce properties for nodes which are never going to use them, e.g. `maxVels` in `CuiAbsolute`. Also, at some point in the future we might develop a motor drive node (analogous to `TechnosoftIpos`) which expects a different set of properties. Therefore, a plausible solution is to segregate those properties per CAN node type and place them in separate groups within the .ini file, e.g. `[TechnosoftIpos_2], [CuiAbsolute_102] etc.`. The CanBusControlboard wrapper should not need to parse and pass them along to wrapped subdevices, just take care of loading the .ini group on a per-subdevice initialization basis.

PeterBowman, 2019-07-31:
Following the .ini rework proposal, it might be wise to parse an additional per-device type group (e.g. all Cuis, all iPOS nodes, etc.) to gather common configuration parameters within (e.g. PT/PVT mode period).

Besides, I'd like to take advantage of the .ini file transclusion feature: let's put all iPOS/Cuis/grippers on their own separate .ini file and transclude them if needed from the parent .ini holding the configuration of the CanBusControlboard master device.

PeterBowman, 2019-09-10:
@jgvictores suggested we could start a `TechnosoftIposWithCui` device that inherits from `TechnosoftIpos` and adds the encoder functionality. I'm quite positive about that, although composition makes a little more sense to me (instantiate a `CuiAbsolute` **within** `TechnosoftIpos`). This will render the role separation mostly pointless if we consider that grippers are less-sophisticated actuators (which does not need to be necessarily true).

PeterBowman, 2019-09-14:
Perhaps the absolute encoders could be enabled by the adequate YARP interface? Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/157 suggests [`yarp::dev::IMotorEncoders`](http://www.yarp.it/classyarp_1_1dev_1_1IMotorEncoders.html#ab348098a114dc663111a7b2e7ea0c414) could fit the relative encoders (uses encoder counts as input/output parameters). So, should [`yarp::dev::IEncoders`](http://www.yarp.it/classyarp_1_1dev_1_1IEncoders.html) be aimed for use with absolute encoders (and fall back to relative encs)?

PeterBowman, 2019-09-17:
Had a brief and pleasant face-to-face with @PeterBowman, some new ideas have been laid out:

* YARP controlboard wrappers should not need to know a thing about internal CAN-to-motor-id associations, this mapping shall be handled by the CanBusControlboard internally and only said motor ids will be exposed. That is, additional CAN nodes which do not perform any movement, for instance the relative/absolute encoders, do not participate in the motor id assignment on the YARP-exposed side. This is a strong argument in favor of embedding everything encoder-related within the motor device, be it the TechnosoftIpos or some kind of gripper we do not have yet. To sum up, CAN node assignment intricacies are orquestated by the CanBusControlboard, whereas internal implementation, including the aforementioned class composition strategy, is meant to be cared for by the raw subdevices.

* There should be nothing special about allowing more than one motor id par CAN node id (https://github.com/roboticslab-uc3m/yarp-devices/issues/171). Some Technosoft drives might already work this way ([ref](https://www.technosoftmotion.com/en/intelligent-drives-and-motors/ipos-line/ipos-multi-axis)). In consequence, motor devices and gripper-like devices would be treated equally in this regard.

* Motor-id-to-YARP-id association regarding gripper devices might benefit from the usual controlboardwrapper mechanisms. We "split" all four TEO's CAN buses in a way that a separate YARP port is opened for the head and trunk body parts. However, the same port is shared by motor drives and grippers (LacqueyFetch, TextilesHand, and the WIP Dextra device (https://github.com/roboticslab-uc3m/yarp-devices/issues/227)). I would like to instantiate separate YARP wrappers for the gripper nodes; then, apart from the usual motor interfaces, additional configuration could be shared on the YARP network with the remote variables interface: https://github.com/roboticslab-uc3m/yarp-devices/issues/228.

PeterBowman, 2019-09-19:
Cui encoders ready at https://github.com/roboticslab-uc3m/yarp-devices/commit/a123bc981bbb7153b79874621065603bb6fbd3e5. Currently no data is requested from the relative encoders, this will be done at https://github.com/roboticslab-uc3m/yarp-devices/issues/232 via TPDO configuration.

PeterBowman, 2019-09-19:
Next (and probably last) big step here: implement some sort of LUT within CanBusControlboard to map exposed joint indexes (e.g. like the `j` in `positionMove(j, ref)`) to motor ids controlled by raw subdevices. For instance, if we open three raw devices A, B and C, where A controls two joints, B three, and C one joint respectively, then `CanBusControlboard::positionMove(4)` should map to `B::positionMoveRaw(2)`. For comparison, check the subdevice mechanism at [ControlBoardWrapper](https://github.com/robotology/yarp/tree/bfca0db72d5592e349aa93445459109a6a7df4ba/src/libYARP_dev/src/devices/ControlBoardWrapper).

jgvictores, 2019-09-20:
Essentially https://github.com/roboticslab-uc3m/yarp-devices/issues/171 originally called "Maybe `raw` index 0 wasn't a great idea"?
Lots has changed, by the main idea of the issue is that the first argument of "(e.g. like the `j` in positionMoveRaw(`j`, ref))" (note the `Raw`) can be exploited to pass useful info.

PeterBowman, 2019-09-20:
Good catch! Yes, https://github.com/roboticslab-uc3m/yarp-devices/issues/171 is definitely the right place for speaking about device mapping. I'll update the status in that ticket.

So, the last thing left to do here: similarly to the process undergone at the CuiAbsolute device, let's remove unused interfaces in LacqueryFetch, TextilesHand and so on. The following is still valid:

> CanBusControlboard should not crash if some of these subdevices don't inherit an unrelated interface.

> What is IInteractionMode and why do we need it? Drop unused interfaces.

PeterBowman, 2019-09-20:
BTW [teo-developer-manual/cui-absolute-values.md](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/6b76e224721dc03b660df89e7269403785be7035/cui-absolute-values.md) should die.

PeterBowman, 2019-09-21:
Summary & roadmap:

- [CuiAbsolute](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/YarpPlugins/CuiAbsolute): done at https://github.com/roboticslab-uc3m/yarp-devices/commit/a123bc981bbb7153b79874621065603bb6fbd3e5.

- [DextraCanControlboard](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/YarpPlugins/DextraCanControlboard): inherits from [DextraRawControlboardLib](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/DextraRawControlboardLib), which takes care of properly implementing motor interfaces.

- [FakeJoint](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/YarpPlugins/FakeJoint): I'm going to rewrite this, see https://github.com/roboticslab-uc3m/yarp-devices/issues/192.

- [**LacqueyFetch**](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/YarpPlugins/LacqueyFetch): most interfaces will be replaced by [`yarp::dev::IPWMControl`](http://www.yarp.it/classyarp_1_1dev_1_1IPWMControl.html).

- [**TechnosoftIpos**](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/YarpPlugins/TechnosoftIpos): just going to remove the `IInteractionMode` implementation.

- [TextilesHand](https://github.com/roboticslab-uc3m/yarp-devices/tree/a9d0ed14a33f14326d96ef13bacff17197c71307/libraries/YarpPlugins/TextilesHand): will undergo some major changes at https://github.com/roboticslab-uc3m/yarp-devices/issues/234, not touching this right now.

Currently working on bolded entries.

PeterBowman, 2019-09-21:
Device-side ready at https://github.com/roboticslab-uc3m/yarp-devices/commit/cfd8d88e8555b8fd399e4ab80db95df9082e138f, the Lacquey firmware update is targeted at https://github.com/roboticslab-uc3m/yarp-devices/issues/235.

closed 2019-12-31 by PeterBowman