roboticslab-uc3m/yarp-devices #213: Issue with joint limit

rsantos88, 2019-05-21:
Doing some tests with the robot, I've noticed the following problem:
The right frontal wrist has now the joint limits between [-115.1º and 44.7º]. This can be confirmed with:
```bash
>>get llimb 5
Response: [is] llimb -115.1 44.7 [tsta] 1 1558445604.35259 [ok]
```
and  has been a joint in which the change of direction of rotation has had to be made.
The problem is, when I try to move the joint to -50º, (front forward movement) the joint always stops suddenly, applying a bad joint limit that is not correct. This it checked with:
```bash
>>set pos 5 50
Response: [ok]
>>get enc 5
Response: [is] enc 48.9279479980469 [tsta] 2 1558445623.22604 [ok]
```
is as if the configured limit were: [-44.7 115.1] 
However, I don't have this problem with left frontal wrist (not changed by DOF)

Related: 
* last [changes](https://github.com/roboticslab-uc3m/yarp-devices/commit/304b2455538479d21e2ecfc77678e596f359353d) of DOF in yarp-devices 
* current [ini](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/develop/share/launchManipulation/conf/launchManipulation.ini) file of launchManipulation

PeterBowman, 2019-05-21:
Only to discard a copy-paste error:

* `get llimb 5`: `llim`, right?
* `set pos 5 50` or `set pos 5 -50`? Front forward should be `-50`, I guess.

rsantos88, 2019-05-21:
> * `get llimb 5`: `llim`, right? 

llim, yes (sorry)

> * `set pos 5 50` or `set pos 5 -50`?

set pos 5 -50

commit 9600e497 (2019-05-22): fixed joint limits problem

Closes #213.
closed 2019-05-22 by rsantos88