roboticslab-uc3m/yarp-devices #224: Display current iPOS drive status in GUI app

PeterBowman, 2019-08-01:
Messages flowing through the CAN network are printed to stdout by CanBusControlboard master+slave devices. This is hard or nearly impossible to follow during execution, not to mention that the `launchManipulation`-like scripts run on the robot's CPUs. In the ideal scenario, we'll use `yarprun` and not even take a glimpse into this output. However, we certainly want to check the state of the drives in an online manner without being forced to plug the CAN/USB/RS232 converter and resort to using EasySetup (which only allows to load a single node from only one CAN bus).

In https://github.com/roboticslab-uc3m/yarp-devices/issues/223, a new CAN application layer implementation will be proposed. I'd like to publish current status word, EMCY messages and so on via YARP ports, either by creating a separate publisher port or using the `yarp::dev::IRemoteVariables` interface (mind its shortcomings, see https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-495322968). Data will be presented to users in a GUI application, which is the point of this issue. Possible implementations:

* Python + PySide2 ([grabberControls2Gui](https://github.com/roboticslab-uc3m/yarp-devices/tree/6f5c2da92bbf20cda288ff62de88cdf530355926/programs/grabberControls2Gui))
* Python + Tkinter ([sensorReader.py](https://github.com/roboticslab-uc3m/yarp-devices/blob/6f5c2da92bbf20cda288ff62de88cdf530355926/programs/sensorReader/sensorReader.py))
* C++ + Qt (much like `yarpmanager`)

jgvictores, 2019-08-01:
Well, there is the YARP log thingy...
- https://github.com/robotology/yarp/pull/252
- https://www.yarp.it/yarplogger.html no pic, so: https://github.com/robotology/yarp/issues/2063 (note on the top right side there are icons to show/hide warning/error/...) (kind of an assumption, never have used it, have used its ROS equivalent that kinda works like that)

PeterBowman, 2019-08-01:
> Well, there is the YARP log thingy...

We have https://github.com/roboticslab-uc3m/developer-manual/issues/11, I'll give it a try and report there my findings. Anyway, the goal of this issue is twofold:

* display current drive status, something à la EasySetup
* redirect logs to file, if possible

I was thinking about a "record to file" button within the GUI that would help achieve the latter. It would either export drive registers (with timestamps), debug logs, or both.

PeterBowman, 2019-08-02:
> I was thinking about a "record to file" button within the GUI that would help achieve the latter. It would either export drive registers (with timestamps), debug logs, or both.

If we choose to adopt yarplogger, this step could be spared. So, let's focus here on broadcasting drive state, only. On a second thought, I'd rather do this in a per-CAN bus manner (CanBusControlboard) instead of the YARP-wrapper way (IRemoteVariables interface). Moreover, I'm only interested in one-way comms, so the RPC model followed by IRemoteVariables is of little use here. Therefore, I'm prone to open a (buffered) port on each CanBusControlboard instance and perform cyclic/on-event writes to remote listeners.

I recall having spoken with @smcdiaz about some means to watch the internal drive state without resorting to EasySetup. I think he meant to attach an interactive display to the robot's back side so that every relevant drive status information flowing through the CAN buses is shown, no intermediaries involved. I know the scope of this issue is different and relies on having a CAN master launched on a running on-board CPU; perhaps it's sufficient for our needs, though, or maybe both solutions can complement each other? Is there any crucial piece of data I have not considered that could be displayed via this proposed GUI app?

PeterBowman, 2020-01-28:
Every single message flowing through the CAN bus is now dumped via YARP port thanks to https://github.com/roboticslab-uc3m/yarp-devices/issues/160. Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/231 provided a means to estimate and stream actual CAN bus load.

> If we choose to adopt yarplogger, this step could be spared. So, let's focus here on broadcasting drive state, only.

Besides the EasySetup-like registers, I'd want this app to display dumped CAN messages on-screen, to enable a user interface for custom commands and SDO-specific remote services, and it also would be nice to include a small yarpscope-like widget to dynamically represent current bus load.

PeterBowman, 2023-01-16:
I'm closing this as too complex and not much worth the effort as I initially thought.

> Besides the EasySetup-like registers

Bit transitions are hard or impossible to spot on a GUI widget due to refresh rates. A fully fledged tool of this kind would provide watch utilities or on-change events. For now, I'm happy with manual debugging via RCP (`/sdo:s` port).

> I'd want this app to display dumped CAN messages on-screen

For this matter, [dumpCanBus](https://github.com/roboticslab-uc3m/yarp-devices/tree/5c89ce63feb607863d36cb8be3b9ee486ccc18ff/programs/dumpCanBus) is your friend (via `/dump:o` port).

> to enable a user interface for custom commands and SDO-specific remote services

This is, once again, `/sdo:s` for RPC-like interactions and `/send:i` for one-way streaming.

> and it also would be nice to include a small yarpscope-like widget to dynamically represent current bus load.

Just use `yarpscope` (https://github.com/roboticslab-uc3m/yarp-devices/issues/231#issuecomment-578864852, via `/load:o` port).

closed 2023-01-16 by PeterBowman