roboticslab-uc3m/yarp-devices #232: Leverage CANopen & general improvements

PeterBowman, 2019-09-17:
Project [[CAN-TEO]](https://github.com/roboticslab-uc3m/yarp-devices/projects/2) gathers several tickets which aim to take full advantage of CANopen protocols with a brand new application layer (https://github.com/roboticslab-uc3m/yarp-devices/issues/223). Apart from the usual refactorization and reduction of code bloat, it is expected to achieve new functionalities and improve existing ones. This ticket tracks said improvements.

PeterBowman, 2019-09-17:
The SDO protocol is a confirmed service, that is, any message we send to a CAN node generates a response. CANopen docs use the terms request/response for CAN reads, and indication/confirm for CAN writes.

In the develop branch, CAN reads (from the CAN master POV) employ a tiny hardcoded delay hoping to receive the expected response before it elapses. I noticed that this delay is too small for certain operations, perhaps always. On the other hand, CAN writes do never expect a confirm message.

Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/223 adds the new `SdoClient` class (name is not final). It encapsulates reads and writes in a user-friendly manner so that there is no need to worry about command specifiers, data lengths and such. Issue Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/159 implements a semaphore-based wait-with-timeout mechanism that turns the hardcoded delays obsolete. With these improvements, all SDO transfers are checked for the incoming response/indication message. In case a timeout occurs, or a SDO abort transfer message arrives (previously not implemented at all), relevant methods return a false boolean.

PeterBowman, 2019-09-17:
CanBusControlboard has always inherited from a `yarp::os::Thread` class to implement a CAN read thread. In https://github.com/roboticslab-uc3m/yarp-devices/issues/209, this has been moved to a separate class and replicated so that CAN writes are orquestated from their own thread, too. This aimed to move all read/write operations to the CanBusControlboard wrapper, and to make the raw subdevices focus on interpreting incoming messages (read thread) and registering the outgoing ones (write thread). Both cases involve the usage of CAN RX/TX buffers for the first time (PeakCAN-software side). The latter case exploits this fact extensively: subdevices place message data in available consecutive slots within said buffer; then, the write thread checks periodically for awaiting messages and send them in one batch.

Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/226 will allow multiple read/write threads, one per CAN bus.

PeterBowman, 2019-09-18:
The ControlBoardWrapper class queries the following data on each `RateThread::run` iteration ([ref1](https://github.com/robotology/yarp/blob/bfca0db72d5592e349aa93445459109a6a7df4ba/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L894-L896), [ref2](https://github.com/robotology/yarp/blob/bfca0db72d5592e349aa93445459109a6a7df4ba/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L930-L938)):

* `IEncodersTimed::getEncodersTimed`
* `IEncoders::getEncoderSpeeds`
* `ITorqueControl::getTorques`
* `IEncoders::getEncoderAccelerations`
* **`IMotorEncoders::getMotorEncoders`** (https://github.com/roboticslab-uc3m/yarp-devices/issues/157)
* `IMotorEncoders::getMotorEncoderSpeeds`
* `IMotorEncoders::getMotorEncoderAccelerations`
* `IPWMControl::getDutyCycles`
* **`ICurrentControl::getCurrents`**
* **`IControlMode::getControlModes`**
* `IInteractionMode::getInteractionModes`

The only ones that entail a CAN network transfer and do not derive from the others are bolded. Both motor encoder reads and actual currents can fit a single TPDO frame. I'm yet to decide on whether control modes should either be handled via local variable or await a TPDO event.

PeterBowman, 2019-10-01:
Control modes YARP queries are no longer forwarded to the CAN network, that is, no CAN request is generated upon `getControlMode()`. Now, control mode changes are listened upon and stored in an internal variable via event callback. See https://github.com/roboticslab-uc3m/yarp-devices/issues/215#issuecomment-537137651.

PeterBowman, 2019-10-02:
According to iPOS CAN user manual (2019), these are the relevant dictionary objects we (may) want to listen to via TPDO:

* 6041h, *Statusword*: UNSIGNED16. Defines current drive state (among other relevant bits), already handled.
* 6061h, *Modes of Operation Display*: INTEGER8. Already handled.
* 1002h: *Manufacturer Status Register*: UNSIGNED32 (comprises object 6041h, *Statusword*). Most importantly, reports I2t warning status.
* 2000h, *Motion Error Register* (MER): UNSIGNED16. Defines most contingencies leading to a fault state.
* 2002h, *Detailed Error Register* (DER): UNSIGNED16. Complements MER with additional error info.
* 2009h, *Detailed Error Register 2* (DER2): UNSIGNED16. Displays detailed information linked to the command feedback error bit in MER. Allegedly available in F514x, only.
* 2003h: *Communication Error Register* (CER): UNSIGNED16. Expands on CAN/SPI/SCI communication errors.
* From section 5.5, *Digital I/O control and status objects*: 60FDh, 208F, 60FEh, 2090h, 2045h, 2102h, 2046h, 2047h and 2055h. These objects relate to digital I/O, analogue inputs and brake status. If possible, it would be nice to leave some PDO room even if we are not going to use them.
* From that same section, see object 2058h, *Drive Temperature*. ~~All I want for Christmas is a temperature sensor.~~
* 6064h, *Position actual value*: INTEGER32. Provides position measurements in user-defined units, which boils down to encoder counts. I'm considering to use 6063h, *Position actual internal value* instead:
  > This object represents the actual value of the position measurement device in increments.
* 207Eh, *Current actual value*: INTEGER16. We want this badly. User manual says only available with firmwares F508I/F509I and above; seemingly, it works on our hardware, though. Object 6077h, *Torque actual value* reflects the same value as confirmed by Technosoft support.

Certain error registers are forwarded from the drives within an EMCY message in case a fault condition arises, see section 4.1.4.1, *Emergency message structures*:

* 2003h, *Communication Error Register* (CER): on 0x7500, "Communication error"
* 2009h, *Detailed Error Register 2* (DER2): on 0x7300, "Sensor error"
* 2072h, *Interpolated position status*: on 0xFF01, "Generic interpolated position mode error (PVT / PT error)"

PeterBowman, 2019-10-03:
Technosoft support has confirmed that object 1002h, *Manufacturer Status Register* can be indeed mapped to a PDO.

PeterBowman, 2019-10-03:
I'm going to remove the iPOS-product-code-to-drive-peak-current mappings so that this value is parsed from .ini file. For future reference (using amperes):

https://github.com/roboticslab-uc3m/yarp-devices/blob/f71f070ffd1da66e76393f2faeecbe12188aee6f/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L19-L66

jgvictores, 2019-10-03:
Makes sense!

PeterBowman, 2019-10-03:
Proposed transmit PDO mappings:

* **TPDO1**: 1002h, *Manufacturer Status Register*, MSR (**4 bytes**; includes 6041h, *Statusword* in the LSB half) + 6061h, *Modes of Operation Display* (**1 byte**) = **5 bytes**

* **TPDO2**: 2000h, *Motion Error Register*, MER (**2 bytes**) + 2002h, *Detailed Error Register*, DER (**2 bytes**) = **4 bytes**

* **TPDO3**: 6064h/6063h, *Position actual (internal) value* (**4 bytes**) + 6077h/207Eh, *Torque/Current actual value* (**2 bytes**) = **6 bytes**

TPDO3 is meant for time critical data streams. For now, I will set a tiny event timer value in async mode and no inhibit time in order to simulate the final desired behavior in synchronous operation (via SYNC signal). Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/223 will expand on that matter. There is room for two additional bytes which could fit the output of a temperature sensor (wink wink @smcdiaz) stream via 2058h.

TPDO1 and TPDO2 are both event driven PDOs. I feel that grouping status-related stuff together is probably more efficient given that such events are prone to fire at any time, as opposed to casual emergency messages linked to fault states that hopefully don't arise that often. Hence, TPDO1 will host statusword and the MSB half of MSR - these are also state-related bits - as well as current mode of operation. On TPDO2, MER/DER should fire on special events and both provide related information.

We are probably fine in reusing the default TPDO configuration, i.e. no event timer and a 30 ms inhibit time. That is, if and only if a bit changes in either PDO, it will produce a message at least 30 milliseconds away from the previous one. In case I decide to set an event timer of e.g. 1 second, those PDOs will send data with such a period, and more often if inhibit time is set as well.

PeterBowman, 2019-10-04:
Ready at https://github.com/roboticslab-uc3m/yarp-devices/commit/b31442bfe11877c32cad700ef628aa87a7a4d7a5, I'm going to revisit this issue after SYNC is implemented and applied to TPDO3 (https://github.com/roboticslab-uc3m/yarp-devices/issues/223).

PeterBowman, 2019-12-25:
> From that same section, see object 2058h, _Drive Temperature_: UNSIGNED16. ~All I want for Christmas is a temperature sensor.~

Oh.

> The ControlBoardWrapper class queries the following data on each `RateThread::run` iteration [...]

(https://github.com/roboticslab-uc3m/yarp-devices/issues/232#issuecomment-532871260) I should mark `IMotorEncoders::getMotorEncoderSpeeds` with bold type due to https://github.com/roboticslab-uc3m/yarp-devices/issues/189#issuecomment-568929803. Thus, room would be made for additional 4 bytes for Object 6069h, "Velocity sensor actual value" or Object 606Ch, "Velocity actual value" (https://github.com/roboticslab-uc3m/yarp-devices/issues/232#issuecomment-538049683).

PeterBowman, 2019-12-25:
Also, there is interesting to note that iPOS firmware provides a low pass filter that can be applied on a 16bit variable value, see Object 2108h. By default, this is configured to filter motor current and can be mapped to a PDO.

PeterBowman, 2019-12-30:
It is important to note that the default inhibit time (30 milliseconds) causes some crucial messages to be lost, e.g. this may occur when modes of operation change twice before this time elapses. It happened to me upon switching from idle mode to position control mode. I did process the statusword callback correctly, but another message was supposed to be generated due to the mode transition, which got lost in the meantime (I presume it was attempted to be broadcast before those 30 milliseconds elapsed).

PeterBowman, 2020-01-02:
Drives are currently configures to stream jont state (encoder reads & current measurements) via TPDO3. This PDO is event-driven, no event timer, default inhibit time (for now) of 30 milliseconds. It means that any new encoder read or current measurement will be broadcast from the drive no sooner than 30 milliseconds after the last message sent. This copes well with our buffering mechanisms, and it is convenient to remember that encoder reads are highly unstable, i.e. the drives are steadily reporting minor variations (hence, in the practice, TPDO3 will stream data at the inhibit time rate despite the motor not being commanded at all).

A question arises: is it convenient to SYNC this particular TPDO? If we take a look at how those reads and measurements are consumed, we notice that the only caller is controlboardwrapper2's periodic thread. Even if we achieve synchronous data streaming across all joints, those data will be consumed at a later time (milliseconds apart). Each drive will report a different acquisition time, so the question boils down to determining whether simultaneousness is a pressing matter here. Client code (YARP-side) will always get the precise instant of data arrival via `IEncodersTimed`. Should this be the same instant across all joints (even across all parts of the robot, in a whole-body *launchCanBus* configuration)?

PeterBowman, 2020-01-08:
> A question arises: is it convenient to SYNC this particular TPDO?

Synchronization helps design a CAN bus that copes well with traffic. Event-driven PDOs make this highly unpredictable. Also, syncing is not as hard as I initially thought.

> A positive answer yields another important question: since all synchronous TPDOs obey the same SYNC signal, and we are sure we want to tell TPDO3 to stream at the SYNC rate, how can other PDOs (RPDO/TPDO) play with this config so that a different rate is used for their specific case, in case of need?

PDOs can be configured in a cyclic synchronous manner, i.e. accept/stream data every 2/3/4... SYNC signals.

PeterBowman, 2020-01-09:
Chapter 6, "Factor group" in iPOS CANopen user manual (2019) looks appealing:

> The iPOS drives family offers the possibility to interchange physical dimensions and sizes into the device internal units.

This set of objects could replace several unit-conversion methods from the [`StateVariables` class](https://github.com/roboticslab-uc3m/yarp-devices/tree/2fb7567908d9133d67158f8e3fdd254f2639d27b/libraries/YarpPlugins/TechnosoftIpos) and move those calculations to the iPOS drives, perhaps with greater precision (see 6.1.10.1, "Setting the numerator and divisor in a factor group object. Example"). Also, object 607Eh, "Polarity" reverses the sign in several position and velocity control-related objects, which is currently performed by our CAN master code via multiplication by 1 or -1 (see `StateVariables::reverse`).

However, I'm not implementing this. There are several places across our codebase that would need a "manual" conversion anyway, that is, the iPOS drives do not apply such unit conversions and sign multiplications in all places we actually need them, hence the helper methods in `StateVariables` would never go away. Also:

> Because the iPOS drives work with Fixed 32 bit numbers (not floating point), some calculation round off errors might occur when using objects 6093 h , 6094 h, 6097 h and 2071 h . If the CANopen master supports handling the scaling calculations on its side, it is recommended to use them instead of using the “Factor” scaling objects.

And:

> In the CiA 402 standard, the dimension and notion index objects have been declared as obsolete.

PeterBowman, 2020-01-09:
TPDO3 configured as cyclic synchronous (every 1 SYNC). The new `syncPeriod` option to CanBusControlboard enables the SYNC timer if present: https://github.com/roboticslab-uc3m/yarp-devices/commit/f7e0cc36ddd44902ddb240eb112793d3b9106a1d + https://github.com/roboticslab-uc3m/teo-configuration-files/commit/c4d354816156778892e8a8737f1b6440c77a67ce + https://github.com/roboticslab-uc3m/asibot-configuration-files/commit/4be82062bbf431778524c1d04e9e69d769909e78. The write message queue is processed as soon as a SYNC message has been registered.

I chose not to communicate CanBusControlboard with its wrapped CAN nodes (and vice versa), possibly via new methods in the `ICanBusSharer` interface, mind the KISS rule. Also, if users need to disable this SYNC timer (unused, e.g. all CAN nodes are fake nodes or not TechnosoftIpos devices), they can comment out the `syncPeriod` in the .ini file.

PeterBowman, 2020-01-09:
Object 2103h, *Number of encoder counts per revolution* reports our `encoderPulses` (4096 in TEO). I'm not going to query this value because of:

> Remark: this object will not indicate a correct number in case a Brushed DC motor is used.

Anyway, dropping the `encoderPulses` value from encoder .ini files is not a good idea.

PeterBowman, 2020-01-16:
Velocity, torque and current commands are now forwarded to the drive via synchronous RPDOs: https://github.com/roboticslab-uc3m/yarp-devices/commit/f4970a267ba8970f54cdd5ef8cfe2607f369c60c.

PeterBowman, 2020-01-19:
> It is important to note that the default inhibit time (30 milliseconds) causes some crucial messages to be lost, e.g. this may occur when modes of operation change twice before this time elapses. It [...]

> Edit2: see [a460407](https://github.com/roboticslab-uc3m/yarp-devices/commit/a460407cd7de60c64f62bacd62134e68895031e2) (attending to motion completion after halt command is issued).

Event timer implemented and defaulted to 250 milliseconds at https://github.com/roboticslab-uc3m/yarp-devices/commit/79a475ebfa0e46a5424a33a1076586fc1fff5bc0 and https://github.com/roboticslab-uc3m/teo-configuration-files/commit/249edbe02f9747fec05611e18c96876aa5d4a1ca.

PeterBowman, 2020-01-24:
Currently, a `get encs` RPC requests yield all-zeroes and a `true` result when drives are powered off, even when not at home position. This has been observed on `launchCanBus --home`.

* Commit https://github.com/roboticslab-uc3m/yarp-devices/commit/3dc753040cfee70090a9c10d5dbedd78107677c9 checks this and reports a failure unless drives have been initialized beforehand.
* Commit https://github.com/roboticslab-uc3m/yarp-devices/commit/f32fbbac3265aff0e8b3e8719799fea3462ad98a prints a warning on failed initial (via `--home`) homing procedure instead of shutting down the process.

Sadly, *yarpmanager* doesn't like a `getEncoders` call yielding `false` and disables certain operations, e.g. it does not start the GUI and, when started, GUI indicators are no longer updated (drive state is not reflected, basically nothing works anymore). Therefore, I reverted the first commit at https://github.com/roboticslab-uc3m/yarp-devices/commit/d59b6af62a168ae06cb788acf09348edb46cbbd0; an upstream PR needs to fix *yarpmotorgui* first.

PeterBowman, 2020-01-25:
> Commit [3dc7530](https://github.com/roboticslab-uc3m/yarp-devices/commit/3dc753040cfee70090a9c10d5dbedd78107677c9) checks this and reports a failure unless drives have been initialized beforehand.

ASWJ we'd better ask the drive for its current "control mode" and test against the `VOCAB_CM_NOT_CONFIGURED` value, which means drive off. Also, we found the [`yarp::dev::IEncoderArrays`](http://www.yarp.it/classyarp_1_1dev_1_1IEncoderArrays.html) interface, part of the multiple analog sensors device group:

> This interface is typically used for group of encoders that are not explicitly controlled by one of the interfaces typically used for motor control, such as IEncoders and IPositionControl, such as encoders measuring the complete state in an underactuated mechanism.

We are good to go with our current setup, that is, treat absolute encoders as tightly coupled with their TechnosoftIpos counterpart per joint (c.f. Dextra hands which actually could pose a nice use case for this MAS interface).

closed 2020-01-27 by PeterBowman