roboticslab-uc3m/yarp-devices #235: Full-range PWM commands in Lacquey firmware

PeterBowman, 2019-09-21:
Commit https://github.com/roboticslab-uc3m/yarp-devices/commit/cfd8d88e8555b8fd399e4ab80db95df9082e138f#diff-68eeea6c71ceb478044ee0f4dc44b6a0 (https://github.com/roboticslab-uc3m/yarp-devices/issues/211) adds a new PWM control layer around the LacqueyFetch device, resembling more closely its internal way of actuating the motor. However, our firmware is severely limited in that only full-positive (+100%), full-negative (-100%) and stop (0%) commands are forwarded, thus ignoring the whole `[-1,1]` value range ([source](https://github.com/roboticslab-uc3m/yarp-devices/blob/8726fa368f8b3becf26a2f263f4d451e1836eb1b/firmware/LacqueyFetch/main.cpp)):

https://github.com/roboticslab-uc3m/yarp-devices/blob/8726fa368f8b3becf26a2f263f4d451e1836eb1b/firmware/LacqueyFetch/main.cpp#L76-L92

Here, `data` is a `char`. Make it a 32-bit `float` so that it matches https://github.com/roboticslab-uc3m/yarp-devices/commit/38176f164de5bd0873013406efda53323c1a415c.

jgvictores, 2019-09-22:
Related:
- https://github.com/roboticslab-uc3m/yarp-devices/issues/234#issuecomment-533788381
- Comment on <http://www.yarp.it/SerialServoBoard_8h_source.html> at https://github.com/roboticslab-uc3m/yarp-devices/issues/234#issuecomment-533858177

\* Not prioritary, just to point out. For starters, SerialServoBoard looks PositionControl-oriented, whereas we both agreed on exposing the more native PWM interface.

jgvictores, 2019-09-22:
Returning to the *scope of the issue*, yes, I totally agree we should recover the full range *at firmware level*.
- This was once available in the `Serial/USB` setup we once had (mentioned at https://github.com/roboticslab-uc3m/yarp-devices/issues/234#issuecomment-533788381).
- Obviously required at firmware level, to have this available via `Canbus` too.

rsantos88, 2019-10-17:
Updated Lacquey firmware:
https://os.mbed.com/users/raulsr988/code/LacqueyFetch/rev/f31b6ddb9cd3/

rsantos88, 2019-10-17:
Note: **Currently the previus firmware is installed on the Mbed.**
Waiting to install the latest changes (38176f164de5bd0873013406efda53323c1a415c) in Teo to be tested completely. Right now the tests have been done under `Pcanview`, sending messages directly to the Mbed. All works correctly. I'll close this issue

PeterBowman, 2019-10-17:
Just one more thing to finally mark this as closed: merge [https://os.mbed.com/users/raulsr988/code/LacqueyFetch/rev/f31b6ddb9cd3/] into [firmware/LacqueyFetch/](https://github.com/roboticslab-uc3m/yarp-devices/tree/fda5803ea68316032b803663f1549db613ef3507/firmware/LacqueyFetch) (also update README).

closed 2019-10-17 by PeterBowman