roboticslab-uc3m/yarp-devices #240: Migrate Jr3 to use YARP Multiple Analog Sensor Interfaces

jgvictores, 2019-12-19:
See http://www.yarp.it/group__dev__iface__multiple__analog.html and [robotology/yarp#1526](https://github.com/robotology/yarp/issues/1526) .

PeterBowman, 2021-11-05:
The `ISixAxisForceTorqueSensors` interface has been implemented at https://github.com/roboticslab-uc3m/yarp-devices/commit/2475bb1f25c45f9b19b0ea27daf4d01bdb81a726 while still preserving `IAnalogSensor` (mapping calls between implementations to avoid code duplication). The device can be still instantiated as an `analogServer`'s subdevice (as usual, via `yarpdev --device Jr3`), but now it can be also wrapped by the new MAS in this way:

`yarpdev --device multipleanalogsensorsserver --subdevice Jr3 --name /jr3 --period 20`

I guess we need to wait for YARP devs to learn whether they decide to keep or nuke `analogServer`.

PeterBowman, 2022-07-17:
The new [ftCompensation app](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/ff5092ac0fd0d9fc4bbe40f2d4678eb4d0d7bf6d/programs/ftCompensation) uses this MAS implementation of Jr3: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/191.

PeterBowman, 2022-07-18:
Our local C++/Python examples have been migrated to using the new MAS interface: https://github.com/roboticslab-uc3m/yarp-devices/commit/de914de92fd227d9675b03a132b15e769bf26bbf. Keep in mind there is no calibration (set-to-zero) method, therefore either restart the device or apply an offset programatically to achieve that.