roboticslab-uc3m/yarp-devices #244: Compare grabberControls2Gui vs iCub frameGrabberGui2-qt and focusTool

jgvictores, 2020-01-13:
Compare our [grabberControls2Gui](https://github.com/roboticslab-uc3m/yarp-devices/tree/d63feba95cd9c44d5a5adc30a2cc7bbb3a66a3e5/programs/grabberControls2Gui) vs iCub [frameGrabberGui2-qt](https://github.com/robotology/icub-main/tree/72f4d060457f8d4e150546e6735dc161faeb9c68/src/tools/frameGrabberGui2-qt) and [focusTool](https://github.com/robotology/icub-main/tree/72f4d060457f8d4e150546e6735dc161faeb9c68/src/tools/focusTool). Discovered via https://github.com/robotology/icub-main/issues/416.

jgvictores, 2020-02-08:
> [focusTool](https://github.com/robotology/icub-main/tree/72f4d060457f8d4e150546e6735dc161faeb9c68/src/tools/focusTool)

Made quite a few hacks to make it compile, modified `main()` to `config.put("device","remote_grabber");`. Segfaults on connect, do not understand intention. Resolution: discarded a.k.a. nevermind. : )

> [frameGrabberGui2-qt](https://github.com/robotology/icub-main/tree/72f4d060457f8d4e150546e6735dc161faeb9c68/src/tools/frameGrabberGui2-qt)

Does seem interesting. I had an older version and was able to use the slider to command a `gain` modification of [YarpOpenraveGrabber](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/develop/libraries/YarpPlugins/YarpOpenraveGrabber). Now it reports `Remote camera device has no IFrameGrabberControl2 interface`. Probably unrelated is the fact that  https://github.com/robotology/yarp/releases/tag/v3.0.0 states "deprecate `IFrameGrabberControl2` in favour of `IFrameGrabberControl`": it's probably just many missing implementations on the `YarpOpenraveGrabbe` side at time of writing.

jgvictores, 2020-02-08:
> Remote camera device has no IFrameGrabberControl2 interface

Seems fixable: https://github.com/robotology/icub-main/blob/6c61f6c507a711962a4f4a961753732ca6e238cd/src/tools/frameGrabberGui2-qt/mainwindow.cpp#L362-L373

jgvictores, 2020-02-12:
> Remote camera device has no IFrameGrabberControl2 interface

Fixed via https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/72dab2af8da703bfd375b1c748deb561de67753f for simulated camera. Still back to situation where only GAIN is controllable.

PeterBowman, 2020-08-18:
> frameGrabberGui2

Note it has been placed on the *Maybe* list for inclusion in a future YARP 3.5.0 release ([ref](https://github.com/robotology/yarp/projects/22)).