roboticslab-uc3m/yarp-devices #249: Failed to assign different joint acceleration in python scripts

rsantos88, 2020-05-29:
I am trying to modify joint speeds and accelerations through python.
I've successfully modified the speed with these lines:
```python
# configure speed
sp = yarp.DVector(axesRA, 5)
posRA.setRefSpeeds(sp)
posLA.setRefSpeeds(sp)
```
I've checked that the joint speed has changed correctly (with `get vels`)... but when I try to modify the acceleration with these lines:
```python
# Configure acceleration 
acc = yarp.DVector(axesRA, 5)
posRA.setRefAccelerations(acc)
posLA.setRefAccelerations(acc)
```
I get the following error message:
```bash
Traceback (most recent call last):
  File "teo-waving.py", line 84, in <module>
    posRA.setRefAccelerations(acc)
  File "/usr/local/lib/python2.7/dist-packages/yarp.py", line 9894, in setRefAccelerations
    return _yarp.IPositionControl_setRefAccelerations(self, *args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'IPositionControl_setRefAccelerations'.
  Possible C/C++ prototypes are:
    yarp::dev::IPositionControl::setRefAccelerations(double const *)
    yarp::dev::IPositionControl::setRefAccelerations(int const,int const *,double const *)
```
Any suggestion? am I calling the function wrong?

PeterBowman, 2020-05-29:
I think the Python bindings are faulty. You can verify this on your *yarp/build/lib/python/yarp.py*, mine poses the following differences between both methods:

```python
def setRefSpeeds(self, *args):
        """
        setRefSpeeds(IPositionControl self, double const * spds) -> bool
        setRefSpeeds(IPositionControl self, int const n_joint, int const * joints, double const * spds) -> bool
        setRefSpeeds(IPositionControl self, DVector data) -> bool
        """
        return _yarp.IPositionControl_setRefSpeeds(self, *args)
```

```python
def setRefAccelerations(self, *args):
        """
        setRefAccelerations(IPositionControl self, double const * accs) -> bool
        setRefAccelerations(IPositionControl self, int const n_joint, int const * joints, double const * accs) -> bool
        """
        return _yarp.IPositionControl_setRefAccelerations(self, *args)
```

Could you just use individual `setRefAcceleration(joint, value)` calls until this is fixed?

jgvictores, 2020-06-01:
@PeterBowman Good catch!

I'd leave this issue open then until we solve it upstream, which should take a while.

@rsantos88 I'm confident you can implement the individual calls, right? :tada:

rsantos88, 2020-06-01:
> @rsantos88 I'm confident you can implement the individual calls, right? :tada: 

Yes, of course. Thanks!

PeterBowman, 2020-06-26:
This is actually https://github.com/robotology/yarp/issues/1966. I'll submit a patch soon.

PeterBowman, 2020-07-12:
WIP at https://github.com/robotology/yarp/pull/2316.

PeterBowman, 2020-07-15:
The fix has been merged into master and is expected for the YARP 3.4 release later this month.

closed 2020-07-15 by PeterBowman