roboticslab-uc3m/yarp-devices #250: Probems regarding the change to velocity mode

rsantos88, 2020-06-11:
When performing experiments with the soft neck, using the new drives (iPOS 3604Mx) and PCB, I had encountered some issues:
When using position mode it works perfectly, and performs the expected trajectory. However, when I tried to change the drivers mode to velocity mode error messages appeared. I tried to change a single driver (driver 0) by rpc:
```bash
>>set icmd cmod 0 vel
Response: [fail]
```
and this is the output obtained by the launchCanBus: [launchCanBus.txt](https://github.com/roboticslab-uc3m/yarp-devices/files/4764365/launchCanBus.txt)
Any suggestions on what could be the problem?

PeterBowman, 2020-06-11:
This is the actual symptom:

`[warning] TechnosoftIpos.cpp:440 handleEmcy(): A wrong mode is set in object 6060h , modes of operation (canId 1)`

I suspect your iPOS drives are not properly configured to operate in velocity mode. Have you ever successfully switched to it before? TEO's drive config resembles the following:

[image]

The "Close position, speed and current loop" checkbox is probably relevant. Please verify it is checked in your iPOS config.

rsantos88, 2020-06-11:
It seems that the problem is going to be that ... I'll tell you the results soon. Thank you!

npuente, 2020-06-12:
As PeterBowman suggested, the problem was related to the drivers configuration. The selected control mode was position mode however, only the current and position loops were enabled. By selecting also the speed loop, the problem was partially solved, as the driver needed more readjusting.
When using launchCanbus, the drivers succeeded in controlling the motors position, but producing an abnormally loud noise. I suspect that the problem came from the speed controller configuration. as it was configured after selecting velocity control for the driver.
The driver was once again configured, this time by selecting first position control (with the 3 loops, position, current and speed), and then the parameters for the torque. speed, and position controllers were tuned. in this same order. This measures solved the problem, and the neck works perfectly fine in both position mode and velocity mode. Thank you for all your help !

rsantos88, 2020-06-12:
i close this issue. Thanks!

closed 2020-06-12 by rsantos88