roboticslab-uc3m/yarp-devices #255: Error while opening /usr/local/lib/yarp/yarp_controlboardwrapper2.so

rsantos88, 2021-01-29:
I want to comment on an error that has occurred quite often for a long time. Sometimes, when I go to start launchCanBus, the following message is displayed in the process of its execution: 
```bash
[info] TechnosoftIpos.cpp:337 interpretModesOfOperation(): Profile Position Mode. canId: 20.
[info] TechnosoftIpos.cpp:58 reportBitToggle(): [status] Bit reset: Target reached. (canId: 20)
[info] TechnosoftIpos.cpp:58 reportBitToggle(): [MSR] Bit reset: Target command reached. (canId: 20)
[INFO] |yarp.dev.PolyDriver| Created device <CanBusControlboard>. See C++ class roboticslab::CanBusControlboard for documentation.
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/lib/yarp/yarp_controlboardwrapper2.so:
  /usr/local/lib/yarp/yarp_controlboardwrapper2.so: undefined symbol: _ZN4yarp2os3Log16forward_callbackE
[WARNING] |yarp.os.YarpPluginSettings| Cannot load plugin from shared library (yarp_controlboardwrapper2)
[WARNING] |yarp.os.YarpPluginSettings| (yarp_controlboardwrapper2: cannot open shared object file: No such file or directory)
[ERROR] |yarp.dev.PolyDriver| Could not find device <controlboardwrapper2>
[error] LaunchCanBus.cpp:117 configure(): Wrapper device leftArmPort configuration failure.
[INFO] |yarp.os.RFModule| RFModule failed to open.
```
Normally, when I want to quickly fix this error, I delete the /usr/local/lib/yarp/yarp_controlboardwrapper2.so file and reinstall yarp (sudo make install) to redo the library installation. It may not be the best solution, as it is temporarily fixed but the error reappears again another day. What could it be due to?

PeterBowman, 2021-01-30:
It baffles me that the fix doesn't persist after a reinstalation. I might have seen this before, perhaps in an upstream issue/patch, in which case it might have been fixed in the meantime. I have updated YARP, so let's see if this happens again:

* head: YARP version 3.4.2+13-20210125.9+gitb946661
* manipulation: YARP version 3.3.3+1-20200911.1+git38fddc0
* locomotion: YARP version 3.4.2+13-20210125.9+gitb946661

rsantos88, 2021-01-31:
OK thanks. I'll watch it  :eyes:

rsantos88, 2021-02-01:
Today, starting the manipulation launch, I've found again the same error that I solved by removing the file and reinstalling it. I leave here the sample of the terminal in case it could give some interesting clue. 
[yarp_controlboardwrapper2 error.zip](https://github.com/roboticslab-uc3m/yarp-devices/files/5904936/yarp_controlboardwrapper2.error.zip)

jgvictores, 2021-02-01:
Could be wrong, but really looks like an issue due to multiple installations of YARP.

I'd really completely wipe everything out, specially do a `sudo rm -rf /usr/local/lib/yarp /usr/local/share/yarp` but also lots of `sudo updatedb; sudo locate yarp; sudo locate YARP` to really purge everything (including symbolic links). Finally recompile+reinstall YARP and recompile+reinstall yarp-devices.

PeterBowman, 2021-02-01:
There are a few more error lines in that output:

```
[INFO] |yarp.dev.PolyDriver| Created device <CanBusPeak>. See C++ class roboticslab::CanBusPeak for documentation.
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/lib/yarp/yarp_portmonitor.so:
  /usr/local/lib/yarp/yarp_portmonitor.so: undefined symbol: _ZN4yarp2os3Log16forward_callbackE
[INFO] |yarp.os.Port| Port /teo/can/rightArm/dump:o active at tcp://2.2.2.51:10003/
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/lib/yarp/yarp_portmonitor.so:
  /usr/local/lib/yarp/yarp_portmonitor.so: undefined symbol: _ZN4yarp2os3Log16forward_callbackE
[INFO] |yarp.os.Port| Port /teo/can/rightArm/send:i active at tcp://2.2.2.51:10004/
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/lib/yarp/yarp_portmonitor.so:
  /usr/local/lib/yarp/yarp_portmonitor.so: undefined symbol: _ZN4yarp2os3Log16forward_callbackE
[INFO] |yarp.os.Port| Port /teo/can/rightArm/sdo:s active at tcp://2.2.2.51:10005/
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/lib/yarp/yarp_portmonitor.so:
  /usr/local/lib/yarp/yarp_portmonitor.so: undefined symbol: _ZN4yarp2os3Log16forward_callbackE
[INFO] |yarp.os.Port| Port /teo/can/rightArm/load:o active at tcp://2.2.2.51:10006/
```

Suggestions:
* `yarpdev --list | grep yarp_controlboardwrapper2`, write down the full path; usually this should be ` /usr/local/lib/yarp/yarp_controlboardwrapper2.so`
* `ldd <path/to/yarp_controlboardwrapper2>`
* `nm -D -C <path/to/yarp_controlboardwrapper2> | grep forward`

PeterBowman, 2021-02-01:
I have a suspition YARP 3.4 was once installed on that PC (manipulation), then we rolled back to YARP 3.3.

* In YARP 3.3, there is a static method `yarp::os::Log::forward_callback` ([ref](https://github.com/robotology/yarp/blob/38fddc046f8573b7d03cdec3d48f869f2a142bf0/src/libYARP_os/src/yarp/os/Log.h#L96)).
* In YARP 3.4, it got renamed to `yarp::os::Log::forwardCallback` ([ref](https://github.com/robotology/yarp/blob/yarp-3.4/src/libYARP_os/src/yarp/os/Log.h#L128)).

So I think the `/usr/local/lib/yarp/yarp_controlboardwrapper2.so` is trying to link against `/usr/local/lib/libYARP_os.so.3.3.3`, but there is a `/usr/local/lib/libYARP_os.so.3.4.x` symlinked by `/usr/local/lib/libYARP_os.so` that shadows old symbols. On a second thought, please try this:

* `ldd /usr/local/lib/yarp/yarp_controlboardwrapper2.so`
* `ls -l /usr/local/lib/libYARP_os*`

rsantos88, 2021-02-09:
* [terminal output](https://github.com/roboticslab-uc3m/yarp-devices/files/5952812/launch_yarp_controlboardwrapper2so.zip) (see the error)
* `ldd /usr/local/lib/yarp/yarp_controlboardwrapper2.so`
 ```bash
 teo@manipulation:~$ ldd /usr/local/lib/yarp/yarp_controlboardwrapper2.so
linux-vdso.so.1 =>  (0x00007ffefd9c6000)
libYARP_dev.so.3 => /usr/local/lib/yarp/../libYARP_dev.so.3 (0x00007f9499cf3000)
libYARP_sig.so.3 => /usr/local/lib/yarp/../libYARP_sig.so.3 (0x00007f9499aae000)
libYARP_os.so.3 => /usr/local/lib/yarp/../libYARP_os.so.3 (0x00007f94996f0000)
libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f949930d000)
libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f94990f5000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9498d2b000)
libYARP_math.so.3 => /usr/local/lib/yarp/../libYARP_math.so.3 (0x00007f9498ae5000)
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f94987dc000)
libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f9498583000)
libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9498366000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9498162000)
libedit.so.2 => /usr/lib/x86_64-linux-gnu/libedit.so.2 (0x00007f9497f2a000)
/lib64/ld-linux-x86-64.so.2 (0x00007f949a22e000)
libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f9497d01000)
libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f9497aec000)
```
* `ls -l /usr/local/lib/libYARP_os*`
```bash
teo@manipulation:~$ ls -l /usr/local/lib/libYARP_os*
lrwxrwxrwx 1 root root      15 feb  1 18:38 /usr/local/lib/libYARP_os.so -> libYARP_os.so.3
lrwxrwxrwx 1 root root      19 feb  9 17:31 /usr/local/lib/libYARP_os.so.3 -> libYARP_os.so.3.4.0
-rw-r--r-- 1 root root 1897168 ene 21 13:04 /usr/local/lib/libYARP_os.so.3.3.3
-rw-r--r-- 1 root root 2206816 sep 11 19:16 /usr/local/lib/libYARP_os.so.3.4.0
```

PeterBowman, 2021-02-09:
Yeah, thank you, that's it: `yarp_controlboardwrapper2.so` links against `libYARP_os.so.3` which is a symlink to `libYARP_os.so.3.4.0`, but you must have everything else (i.e. yarp-devices+launchCanBus) compiled against `libYARP_os.so.3.3.3`, so the linker goes crazy.

Please `rm /usr/local/lib/libYARP_*` (do `ls` first and check just in case). Make sure YARP is on the `yarp-3.3` branch, compile and reinstall, then proceed likewise with yarp-devices.

rsantos88, 2021-02-16:
Done! Thanks for your help

closed 2021-02-16 by rsantos88