roboticslab-uc3m/yarp-devices #261: Implement friction compensation

PeterBowman, 2021-06-25:
TEO's joints require a lot of effort to rotate because of friction. It might be useful to have some sort of out-of-the-box compensation for this phenomenon without having to rely on external control loops.

I believe friction compensation should only target torque/current control as position and speed loops already take care of following their reference regardless of any physical downstream reactions (friction, gravity, load...) that could impede motion. Without any deeper background on control theory, it is only my guess that the acceleration feedforward component of the position PI kind of behaves like a term of the intrinsic compensator.

Regarding the friction model, I'd go first with [static friction](https://w.wiki/3YHr) (a.k.a. Coulomb friction). Further, a damping coefficient could be considered in order to apply an additional friction term that varies with the joint's rotation speed. These two friction kinds are represented in Gazebo via `<friction>`  and `<damping>` joint properties ([SDF reference](http://sdformat.org/spec?ver=1.6&elem=joint#axis_dynamics)). Note it seems to go a bit further and even consider a spring-like behavior via the additional initial-reference and stiffness parameters. Refs: [`Joint::Load`](https://github.com/osrf/gazebo/blob/e4400ae7e04e4859eec392ac67b813402703bab0/gazebo/physics/Joint.cc#L155-L177), [`ODEJoint::SetParam`](https://github.com/osrf/gazebo/blob/e4400ae7e04e4859eec392ac67b813402703bab0/gazebo/physics/ode/ODEJoint.cc#L449-L456) (regarding friction), [`ODEJoint::SetStiffnessDamping`](https://github.com/osrf/gazebo/blob/e4400ae7e04e4859eec392ac67b813402703bab0/gazebo/physics/ode/ODEJoint.cc#L1037-L1088) (regarding joint stiffness/damping, [docs](http://osrf-distributions.s3.amazonaws.com/gazebo/api/9.0.0/classgazebo_1_1physics_1_1Joint.html#a0c077cd57f03c5e6517e472df27553b3)).

According to estimations by S. Morante & J. Victores, static friction and damping coefficients on TEO's left sagittal shoulder result in ~0.7 and ~0.001, respectively (https://github.com/roboticslab-uc3m/teo-gazebo-models/issues/1#issuecomment-754173487). See [morante2015force-preprint.pdf](https://github.com/roboticslab-uc3m/yarp-devices/files/6715670/morante2015force-preprint.pdf).

See also an attempt on friction compensation in the iCub: https://github.com/robotology/icub-firmware/issues/67, [`PID_do_friction_comp`](https://github.com/robotology/icub-firmware/blob/4c52093f3d654c1a9d06c41e951c16e4d964f9d1/emBODY/eBcode/arch-arm/embobj/plus/mc/Pid.c#L178-L181).

PeterBowman, 2021-06-28:
Note iPOS drives already implement some sort of compensation by applying a current feedforward. Sadly, I think it doesn't care of the sign of the velocity, hence it might help or hinder motion depending on it. This is also the reason why certain joints (sagittal right hip?) move out of zero pose right after entering torque mode. @jgvictores said this feedforward was disabled for the sake of their tests.

jgvictores, 2021-06-29:
> I think it doesn't care of the sign of the velocity

I confirm this is my belief but I'm not 100% sure.

> This is also the reason why certain joints (sagittal right hip?) move out of zero pose right after entering torque mode

I confirm I'm also pretty sure about this, which provides further proof of the above (the joint does not even wait for a reference to check the direction, it just feeds the current).

> said this feedforward was disabled for the sake of their tests.

I'm quite positive on this one.

PeterBowman, 2021-06-29:
The iPOS setting is available in EasySetup > Drive Setup > Current controller > Tune & Test > Test > Compensate for gravitational load > Offset value. 

[image]

It is described like this in the documentation:

> For applications with gravitational loads you can add an offset to the current reference to compensate the gravitational force. Check “Compensate for gravitational load” and set the “Offset value”. The maximum value of the offset is limited at motor nominal current.

Assuming the latest backup represent the current state of the drives, the only place we are using this parameter is joint 12, left sagittal ankle, with a value of 0.25 A ([variable *OffsetCurrent*](https://github.com/roboticslab-uc3m/teo-ipos-backup/blob/a38a2053a8de42518f1a6d8b71123e00d7b5508f/config/ID-12/parameters.cfg#L178)).

PeterBowman, 2022-03-14:
`yarp::dev::ITorqueControl` now exposes friction parameters, namely Coulomb and viscous friction coefficients, via `yarp::dev::MotorTorqueParameters`: https://github.com/robotology/yarp/pull/2805. We've got this covered at https://github.com/roboticslab-uc3m/yarp-devices/commit/d5367fb901cc8dd7ed0658fd030bd49bd9971cb9, although currently unused.

PeterBowman, 2023-01-17:
Now we have a custom PID implementation: https://github.com/roboticslab-uc3m/yarp-devices/issues/262. Following on this, I'm thinking of a nested (cascaded) torque PID capable of accounting for friction, whose output would be directly forwarded to the iPOS firmware as current commands.

PeterBowman, 2023-02-09:
See https://robotology.github.io/robotology-documentation/doc/html/classiCub_1_1ctrl_1_1OnlineStictionEstimator.html:

> Online Stiction Estimator.
>
> Estimate the up and down stiction values.
> During the experiment, the joint is controlled by a high-level pid controller that commands directly the voltage in order to track a time varying reference position. The stiction values are estimated during the rising and falling edge transitions.

PeterBowman, 2023-08-29:
[`yarp::dev::MotorTorqueParameters`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1MotorTorqueParameters.html) will probably lose its `bemf` and `bemf_scale` members in the near future, whereas a new one regarding velocity thresholds will be added in YARP 3.9 (https://github.com/robotology/yarp/pull/3008).