roboticslab-uc3m/yarp-devices #267: Implement new gripper device: qb SoftHand2 Research

PeterBowman, 2023-09-04:
* https://qbrobotics.com/product/qb-softhand-2-research/
* https://qbrobotics.com/download/qb-softhand2-research-downloads/
* https://wiki.ros.org/Robots/qbhand
* https://bitbucket.org/qbrobotics/qbdevice-api-6.x.x/
* https://bitbucket.org/qbrobotics/qbdevice-api-7.x.x/
* [user manual](https://qbrobotics.com/wp-content/uploads/2023/01/qbSoftHand2-Research-User-manual-GENERAL-4.pdf) ([PDF](https://github.com/roboticslab-uc3m/yarp-devices/files/12513237/qbSoftHand2-Research-User-manual-GENERAL-4.pdf))
* [quick start guide](https://qbrobotics.com/wp-content/uploads/2023/01/qbSoftHand2-Research_quick-start-guide-1.pdf) ([qbSoftHand2-Research_quick-start-guide-1.pdf](https://github.com/roboticslab-uc3m/yarp-devices/files/12513240/qbSoftHand2-Research_quick-start-guide-1.pdf))
* [datasheet](https://qbrobotics.com/wp-content/uploads/2023/01/qbSoftHand2-Research-datasheet-4.pdf) ([PDF](https://github.com/roboticslab-uc3m/yarp-devices/files/12513247/qbSoftHand2-Research-datasheet-4.pdf))

PeterBowman, 2023-12-08:
YARP 3.9 has introduced a new [`IJointCoupling` interface](https://github.com/robotology/yarp/blob/a62aa78553a9afadb1bf9d2d04da40d38d70704d/src/libYARP_dev/src/yarp/dev/IJointCoupling.h). Perhaps not so useful here than on the Dextra hand (ping @jgvictores), but it's interesting anyway.

jgvictores, 2023-12-08:
> YARP 3.9 has introduced a new [`IJointCoupling` interface](https://github.com/robotology/yarp/blob/a62aa78553a9afadb1bf9d2d04da40d38d70704d/src/libYARP_dev/src/yarp/dev/IJointCoupling.h). Perhaps not so useful here than on the Dextra hand (ping @jgvictores), but it's interesting anyway.

If I'm understanding the objective correctly, it's mostly useful for simulation. https://github.com/roboticslab-uc3m/tools/tree/d7f4442d7a9289870ca7dc253fd4c914f435de86/libraries/YarpPlugins/RealToSimControlBoard which is generic could probably benefit from it.

PeterBowman, 2024-02-14:
Introducing ControlBoardCouplingHandler (implements `IJointCoupling`): https://github.com/robotology/yarp/pull/3085.