roboticslab-uc3m/yarp-devices #273: launchCanBus [ERROR] |yarp.devices.controlBoard_nws_yarp|controlBoard_nws_yarp| Encoder timestamps are not consistent! Data will not be published.

jgvictores, 2024-07-29:
Upon launching `launchCanBus`, loop of: `[ERROR] |yarp.devices.controlBoard_nws_yarp|controlBoard_nws_yarp| Encoder timestamps are not consistent! Data will not be published.`

Consequence: Programs such as `yarpmotorgui --from yarpmotorgui/teo.ini` or `bodyExecution --robot /teo` cannot be run, e.g. the first gets stuck in `[ERROR] leftArm iencs->getEncoders() failed, retrying...` loop in this specific case.

jgvictores, 2024-07-29:
Ugly hack:

$ diff manipulation-noHead-demo.ini manipulation-jj.ini 

```diff
< socket-leftArm ("id21-ipos" "id22-ipos" "id23-ipos" "id24-ipos" "fake-AxialLeftWrist" "id26-ipos" "fake-AxialNeck")
< socket-rightArm ("fake-FrontalNeck" "id15-ipos" "id16-ipos" "id17-ipos" "id18-ipos" "fake-AxialRightWrist" "fake-FrontalRightWrist")
---
> socket-leftArm ("id21-ipos" "id22-ipos" "id23-ipos" "fake-FrontalLeftElbow" "id25-ipos" "id26-ipos" "fake-AxialNeck")
> socket-rightArm ("fake-FrontalNeck" "id15-ipos" "id16-ipos" "id17-ipos" "id18-ipos" "id19-ipos" "id20-ipos")
```

And then run `launchCanBus --from manipulation-jj.ini`.

The specific change is:

```diff
id26-ipos
---
fake-FrontalLeftElbow
```

Because id26 was giving problems. `bodyExecution --robot /teo` was the one indicating it specifically required `FrontalLeftElbow`, thus the `fake-FrontalLeftElbow` naming (after a bogus `fake-whatever` attempt).

closed 2024-07-29 by jgvictores

PeterBowman, 2024-07-29:
There is something fishy in your last diff, id26 can't be the left elbow. Anyway, absolute encoders are known to cause initialization issues. Whenever such error shows up, please scroll up and find the root cause. Should it be a faulty encoder (most commonly a timeout occurring during its initialization), please disconnect it physically and comment out the externalEncoder line in the corresponding .ini file of the affected iPOS driver (e.g. id26-ipos.ini).

jgvictores, 2024-07-29:
Yes, you're completely right. I'm not really sure if the process you mention is described elsewhere, so having it here is a good thing, thanks!

PS: Correct, `FrontalLeftElbow` does not look like a good match for `id26`. Note taken that actual `name`<->`id` correspondences are documented at <https://robots.uc3m.es/teo-developer-manual/diagrams.html> and <https://robots.uc3m.es/teo-developer-manual/tables.html>.

PeterBowman, 2024-07-30:
The preliminary checks and the launching of any TEO demo are described in a few places:
- https://robots.uc3m.es/teo-developer-manual/driver-check-procedure.html
- https://robots.uc3m.es/teo-developer-manual/demo-procedure.html
- https://github.com/roboticslab-uc3m/teo-self-presentation?tab=readme-ov-file#usage

jgvictores, 2024-07-30:
Okay! https://robots.uc3m.es/teo-developer-manual/driver-check-procedure.html#troubleshooting is where  `commenting out (with //) the externalEncoder line in the corresponding idxx-ipos.ini file` is documented. Imho we now have an extra goody: Some specific warning/error messages that can appear during `launchCanBus` runtime now link to that solution via this issue. 🎉